MiczFlor/RPi-Jukebox-RFID #392: Sound Volume: classic vs spotify

deathly87, 2018-12-17:
Hallo zusammen,

habe mir ein neues System mit Stretch Lite 2018-11-13 mit der "One Line Command Install" für Classic aufgesetzt und die USB Soundkarte entsprechend eingerichtet und es funktioniert alles einwandfrei.

Jetzt wollte ich aus Interesse das Ganze als Spotify aufsetzen und im Grunde funktioniert es auch, aber überall, wo ich die Lautstärke auf 100 setze, ist es nur halb so laut, wie mit der Classic Version.

Habt Ihr ein Tipp für mich?

Gruß
André

MiczFlor, 2018-12-17:
Hi @deathly87 
Interesting, Mr. Bond... ich kann jetzt aus dem Kopf auch keinen Grund dafür finden. In beiden Fällen sollte volume über mpd gesteuert werden. Hast du auf alsamix im terminal mal geschaut ob das Master Volume aus irgendwelchen Gründen runtergesetzt wurde? (wobei ich nicht mal weiss, ob das damit zusammenhängen würde...)

deathly87, 2018-12-17:
Das war einfach.. es lag wirklich alsamixer, weil der auf 30 stand... hochgeschraubt und nun geht alles, wie gewünscht..

Vielen Dank für das mega Projekt, welche Du hier uns zur Verfügung stellst ;)

MiczFlor, 2018-12-18:
Danke für die Blumen und viel Spaß mit deiner Phoniebox. Nicht vergessen: Bilder posten!!! 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-12-18 by MiczFlor