MiczFlor/RPi-Jukebox-RFID #393: Shopping list buttons

ndd, 2018-12-17:
Unfortunately the wiki doesn't seem to support comments, so I have to misuse the issue tracker, sorry about that :)

I would love to equip my box with GPIO Buttons, but can't find a shopping list.
Could someone provide links to proven items? Especially buttons of course. As well as the suitable cables and the additional board.

Thanks a lot!

tobiasnix, 2018-12-18:
I am using this buttons https://www.amazon.de/gp/product/B075DCB7LT and this cables https://www.amazon.de/dp/B01EV70C78/ Just use the one with female connector for the raspberry and solder the other part to the button. Buttons are not in A++ quality but working.

Wasted90Youth, 2018-12-18:
I got mine from AliExpress. If you have the time to wait about 4 weeks. They are pretty good as well.
Also got almost all the other stuff from there.

https://de.aliexpress.com/item/12-teile-lose-chrom-Beleuchtete-Arcade-Taste-mit-5-v-LED-Lampe-Mikroschalter-f-r-Arcade/32840724833.html?spm=2114.13010708.0.0.3a994c4d9LKZ7Q

ndd, 2018-12-18:
Thanks a lot.

I've already bought this box:
https://www.amazon.de/gp/product/B00WSXXLZS/

The cables look a lot like your cables. Can I use them as well to stick them onto the buttons contacts? For me those contacts look wider... Or a the standard cables fine?

Thanks again.

Ardios, 2018-12-18:
Hey,

auch hier bin ich so dreist und antworte einfach mal auf Deutsch.

Du könntest auch die Taster aus deiner Box nehmen, die funktionieren auch ohne Probleme.

Wenn du allerdings Arcade Taster nutzen möchtest, bräuchtest du entweder andere Kabel (Die Kontakte von den Arcade Tastern sind breiter) oder müsstest löten.

Auf Amazon gibt es aber auch Touch Taster, diese würden auch mit deinen Kabeln funktionieren https://www.amazon.de/AZDelivery-TTP223B-Digitaler-Kapazitiver-Arduino/dp/B07CRGGXBL/

Grüße
Daniel aka Ardios

ndd, 2018-12-19:
Kein Problem, danke für deine Antwort :)

Möchte Arcade Taster nutzen und habe sie auch schon bestellt.
Sind die oben verlinkten Kabel (https://www.amazon.de/dp/B01EV70C78/) dann richtig für die breiteren Kontakte?

aeskimo, 2018-12-19:
HI
gibt es keine kleinen und schöne buttons?

Ardios, 2018-12-19:
@aeskimo

Du kannst eigentlich alles an Buttons nutzen.

@ndd

Bei meinen Tastern waren die entsprechenden Kabel dabei, wie @SpookyFishes schon schrieb, entweder Löten oder Kabelschuhe (Entweder zum Crimpen oder zum Quetschen) hast du die Taster denn schon da?

ndd, 2018-12-20:
Habe die Buttons schon und es waren leider keine kabel dabei.
Werde meine Kabel jetzt auch einfach dran löten.

Danke.

RiekeIcke, 2018-12-27:
Hallo, ich habe mit Flachsteckhülsen gearbeitet, die hatten 2,4mm und sie an das Male-Ende vom Jumperkabel ohne schwarze Plastikhülle direkt rangequetscht.  

Habe auch erst Arcade-Buttons genommen, da haben die Hülsen perfekt gepasst. Bin dann auf kleinere ungeschwenkt und dort hatten die Hülsen ein bisschen Spiel, passen aber immernoch gut!

Hülsen (nur die kleinsten sind relevant): ManYee 150 Stück Weiblich... https://www.amazon.de/dp/B0795QNWKQ?ref=yo_pop_ma_swf

Buttons (kleinere): Cylewet CLW1078... https://www.amazon.de/dp/B074XDQL7Y?ref=yo_pop_ma_swf

c-m0s, 2019-01-01:
Diese [Arcarde-Buttons](https://www.sparkfun.com/products/9336), in Deutschland z.B. via [Eckstein](https://eckstein-shop.de/SparkFun-Concave-puch-button-for-Arcade-machine-games-blue) nehmen zwar viel Platz weg, sind dafür aber irre rubust und machen ein sattes, fühlbares KLICK beim Betätigen. Ich finde die großartig.

closed 2020-01-18 by MiczFlor