MiczFlor/RPi-Jukebox-RFID #395: Kartenmanagement

chriz77, 2018-12-18:
Hallo,

ich habe diverse Ordner mit je einer Conni Folge und eine Karte mit Conni Label. Nun würde ich gerne der Karte immer wieder verschiedene Ordner zuweisen. Irgendwie geht das nicht. 

- Wenn ich auf die Karte gehe und einen anderen Ordner zuweisen möchte kommt die Meldung "This is too much! Please select only one audiofolder. Make up your mind."
- Wenn ich auf die Karte gehe und "Remove card id" klicke passiert nichts.
- Wenn ich unter "Track Management" einen Track in einen anderen Ordner verschieben möchte passiert ebenfalls nichts.

Mache ich was falsch oder Ist das ein Bug (v1.1.8 classic)?

Danke!
Chris

Ardios, 2018-12-18:
Hey,

versuche mal wenn du auf "Register new card ID" klickst, dann scannt du die Karte die du nutzen möchtest (Auch wenn diese schon verlinkt ist) wählst dann den neuen Ordner aus und dann auf speichern. Nun sollte die Karte eigentlich auf den neuen Ordner verweisen...also in dem Sinne musst du es so machen als wenn die Karte noch nicht registriert wäre

chriz77, 2018-12-18:
Danke so geht's!

Aber für was sind die oben erwähnten Optionen, wenn sie nicht funktionieren?

cubertt, 2018-12-21:
Bug, denke ich, ich kann auch keine Karten (ids) löschen...

MiczFlor, 2018-12-23:
Hi @chriz77 @cubertt 
ihr habt recht, da stimmt was nicht. Ich weiss, dass es mal ging, insofern gebe ich mir die Schuld, dass da bei anderen Besserungen irgendwo was eingerissen wurde. Ich schreib da mal Bug an den thread.

gizmo21, 2018-12-23:
jep bei 1.0.0 geht der delete.

MiczFlor, 2019-01-09:
Fixed in `develop` branch

closed 2019-01-09 by MiczFlor