MiczFlor/RPi-Jukebox-RFID #396: Powerbank without "power button"

petschni, 2018-12-18:
I bought the recommended Powerbank (AUKEY Powerbank 20000mAh) but the problem is that you always have to press the power button to start the raspberry. 

I have build a box that contains the powerbank and added an external power switch that I want to use. Are there any recommendations for a powerbank without a dedicated power button?

tobiasnix, 2018-12-18:
Same problem here. I tried several pwerbanks but it seems like today mostly all powerbanks have a power button. My current workaround is to use only the gpio button for power on/off. The raspberry is in "sleep" mode and will still consume a few mAh which will keep the powerbank on.

aeskimo, 2018-12-19:
Vielleicht gibt es einem Möglichkeit den Ladestand über die Py abzurufen.
Im Anschluss bei niedrigen Ladestand den Hinweis Powerbank bitte laden.

cubertt, 2018-12-21:
> Vielleicht gibt es einem Möglichkeit den Ladestand über die Py abzurufen.
> Im Anschluss bei niedrigen Ladestand den Hinweis Powerbank bitte laden.
> 
> http://raspi-ups.appspot.com/en/index.jsp

Needs LAN  Connection!  If there is no LAN --> it thinks that power is off --> it does not measure the remaining voltage or capacity of the powerbank ....
What a bummer.

A good powerbank:

[Intenso Powerbank 7332530 Slim externes Ladegerät (10000mAh, geeignet für Smartphone/Tablet PC/MP3 Player/Digitalkamera) schwarz
](https://www.amazon.de/gp/product/B015CIZEF0/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1
)

The next one even allows "live"charging without reboot (you can plug that charging cable in and out and the raspi does not reboot):

[iMuto Powerbank 10000mAh Mini Externer Akku USB Tragbar Handy Ladegerät LED-Display Smartphone Akkupack Schwarz](https://www.amazon.de/iMuto-Powerbank-Ladegerät-LED-Display-Smartphone-Schwarz/dp/B018R0F4YQ/ref=sr_1_6?s=ce-de&ie=UTF8&qid=1545425837&sr=1-6&keywords=imuto)

this  feature should be addable to any other powerbank with an addition capacitor, just search the phoniebox issues here at github.

Have fun building!

gizmo21, 2018-12-25:
I still have no final solution for shutdown (and complete circuit off after soft-shutdown), but as battery I have a RAVPower 22000mAh (DE RP-PB052(B) https://www.amazon.de/dp/B01FM6TOH6/ ) model which features pass-through-charging (UPS , Charge-through), so charging AND playing music possible and it has a button to check charge status and activate/power the USB-ports.

BUT I tested and soldered a switch into USB-cable by switching the 5V+ line to the PI.
So after soft-shutdown and manually switching-off the PI with my cable, and waiting e.g. 2 days (without any powerdrain), I can switch on the 5V+ powering to the PI and it will switch on the powerbank USB-Ports without having to press the powerbank-button.

So hopefully when I have more time I also find a simple softwate-shutdown&ciruit-off solution and a power-push-button (instead of switch) to finally have a complete OFF/ON solution without powerdrain.

petschni, 2018-12-30:
@cubertt @gizmo21 thanks for the hints - that was helpful!

closed 2018-12-30 by petschni