MiczFlor/RPi-Jukebox-RFID #399: Suddenly jumps to another folder when playing

bjochen, 2018-12-19:
When playing a folder, after some minutes it jumps to a random folder or stream and starts plaing.
Is this a mpd issue and is there a way to rebuild the database or similar?

Thank you very much and thanks for the great project - my two year old has a lot of fun with the phoniebox (well, sometimes he is a bit irritated when the player start shuffeling the songs ;-) )

BR Jochen

Version1.1.8-beta
EditionClassic edition (barebones)

cubertt, 2018-12-21:
Did you try a fresh install? (another sd card for testing).
Never experienced this.

cu

MiczFlor, 2018-12-23:
Hi @bjochen 
I also never experienced this. The latest version is 1.1.8 - finally released :) Trying the one line install on top of the current install should work (no guarantee) and it would be best to upgrade, because the beta version branch will not be fixed.
All the best and merry christmas,
micz

closed 2018-12-23 by MiczFlor

bjochen, 2018-12-26:
Thanks a lot for your replies! Currently, while visiting the parents, all works fine - it might be due to the fact that wifi is not used here. 

At home it was running in a DMZ an was completely open to the internet via a static ip (that was to beeing able to configure while on my way to work - and that specific ip was rechable through a link on a wordpress page for some days) - my guess (novice in linux and things) that from time to time a search bot (or a person) has opened a link (to an album) an the box started playing ;-)

Wishing you nice holidays & BR
bjochen