MiczFlor/RPi-Jukebox-RFID #401: Add wifi config instead of replacing it when using the UI

petschni, 2018-12-20:
If I read the code correctly currently when you enter new wifi it will replace any existing, correct? 

I think it would be a nice enhancement if the new wifi just gets added to the config and does not replace existing ones. I am happy to contribute the code if that is a feature more people would like to have and if it makes sense to you @MiczFlor ?

ndd, 2018-12-20:
> To add multiple wifi networks you can simply add and ssid section into dhcpd.conf. 

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#faq

Do you mean like this?

gizmo21, 2018-12-20:
This is a section I added https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#i-want-to-add-another-wi-ssid-eg-at-a-friend  

On top there is a "hidden" .sh in the frontend to easily add new wifis the the correct files,  just open in browser:

<yourphonieboxip>/userScripts.php
or 
phoniebox/userScripts.php

select the addhotspot.sh and add the 
<newssid> <newpassword> 
in the parameterfield.

atm priority of wifis have to be added manually, but usually only one wifi is at one place so priority is not that important then.

More info here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/userScripts.php

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/userscripts/addhotspot.sh

MiczFlor, 2018-12-23:
Hi @gizmo21 
thanks for pointing to your section in the wiki. I am hesitant to spend too much time on making both options in one. Parsing config files is something I don't like doing, because usually it will go wrong for someone and or with the next OS update, and then it's a neverending bug catching,
@petschni if you want to give this a shot, I would love to see your approach - and you might know a better way to add this AND make it future proof. So please go ahead,
All the best, micz

gizmo21, 2018-12-23:
Hey @MiczFlor didn't want you to do anything with it, you have enough to fix in maincode to get a stable version out.
Was just to show @petschni that an alpha version is already there.

And coincidentally just successfully tested it at the grandparents wifi, by entering the wifi credentials at home with my .sh method and now booted the box at the grandparents and it connected correctly.

btw. In frontend now the new SSID shows with hashed psk password instead of plaintext one.

petschni, 2018-12-30:
@gizmo21 did not know about that option - thanks for pointing it out! It pretty much does what my use case would be. Would love to see that being integrated in the "main ui".

@MiczFlor My idea around the logic was:

```
if(!wpa_supplicant.conf) {
     // use your template and add the data from the wifi
} else if(wpa_supplicant.conf && wifiNameExistsAlready){
    // replace the password with the new value (could cause parsing issues...)
} else {
    // add the new config to the end of the file
}

```

Yordan1976, 2019-01-24:
addhotspot.sh couldn't be executed on a fresh installation.
-> chmod 775 /home/pi/RPi-Jukebox-RFID/scripts/userscripts/addhotspot.sh

sikraemer, 2019-08-13:
Hi,

a friend of my and I recently build a phoniebox. 
At first we were thinking of adding some AP mode to configure the Wifi settings at new locations (Hotels, Grandparents, etc.) but we decided it would be easier to just connect to a AP that we could open on our phones.

Our version now has not only one Wifi but multiples and even provide them with a priority.
Also the interface now shows the network the box is connected to at the moment (helpful when connected via ethernet).

Should I create a pull request for the change or should I rather provide the changes separately on GitHub.

MiczFlor, 2019-08-13:
Hi @sikraemer 
Please create a pull request. And write a few lines hiw it operates so i can invlude it in rhe documentation. 
All the best, micz

sikraemer, 2019-08-18:
Hi @MiczFlor 

Today I found time to create the pull request: #565

How detailed would you like the description?
Technical background or just how it behaves in the GUI?

Kind regards,
 Simon

MiczFlor, 2019-11-24:
Hi @sikraemer 
coming back to your contribution from August: I meant a end user information, what it does and how to make use of it. Sorry for the late reply :)

closed 2020-01-18 by MiczFlor