MiczFlor/RPi-Jukebox-RFID #407: error at installation

ddbkva, 2018-12-26:
hey
sorr i got an error at the installation, i can`t use the HXGCoLtd rfid reader
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-BbGCwX/pi-rc522/
If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)

Have you connected your USB Reader? [Y/n] y
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
ImportError: cannot import name get_devices
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

$ python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
ImportError: cannot import name get_devices

can somebody help me ?

ddbkva, 2018-12-26:
$ ls -la /dev/input/by-id/
total 0
drwxr-xr-x 2 root root  80 Dec 26 10:00 .
drwxr-xr-x 4 root root 140 Dec 26 10:00 ..
lrwxrwxrwx 1 root root   9 Dec 26 10:00 usb-HXGCoLtd_27db-event-kbd -> ../event1
lrwxrwxrwx 1 root root   9 Dec 26 10:00 usb-Logitech_Logitech_USB_Speaker-event-if02 -> ../event0

splitti, 2018-12-26:
I have the same Error, on different Installations.

MiczFlor, 2018-12-26:
Hi @ddbkva @splitti @alex-voigt 
I just merged @alex-voigt script changes into `master`. The RFID reader landscape really doesn't make my life easier :) it keeps coming and I don't have a battery of RPis with all cards at home (nor do I have the time to check it all). It would great to create a "task force" to fix this issue and find the one script to rule them all...
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/409/files