MiczFlor/RPi-Jukebox-RFID #415: Help: Playing small jingles in parallel with audio playback?

mangiari, 2018-12-29:
Hi there,

I'm currently building a quite customized version of the PhonieBox, with GPIO buttons for the playlists instead of rfid input. To keep the device flexible for future use (childs grow so fast), I dropped the idea of having a hardware icon per gpio playlist button, but instead use sound to document which is doing what. So the idea is to have a jingle for each playlist, e.g. if one playlist holds all kind of piano music, the jingle could be the first 10 notes of "Für Elise"...

When hitting one of the playlist buttons, the jingle should be played, without interfering with the current playback. Just additionally on top, maybe bit louder than the current playback (if any).

Is this easily possible on the raspberry at all? Is it easily possible with the used MPD system?

I'm a quite experienced developer with web stuff, java and also microcontroller and electronics stuff, but never did much with python nor raspberries so far. So I would need some hints to start from. I will try to share whatever I achieve in a PhonieBox fork, so that others can also use the results.

MiczFlor, 2018-12-29:
Hi, I marked this as a feature request. It's not trivial to build... but everything is possible :)

mangiari, 2019-01-01:
Thanks for the quick action. Technically it seems quite simple. When I call the console command that is used for the startup sound `/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3` while phoniebox is playing audio, it is played in parallel with the existing audio. So it's just a matter of hacking your code without messing it up too much ;)

I will build something and try to have it an optional feature by name convention or similar. Would you prefer something with a special named file in each and every audio folder, or something registered in a more central location, like the card ids?

comdoxx, 2019-01-12:
Seems a little similar to my Feature request (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/437) apart from the fact, that file specification by GPIO/card is needed

mangiari, 2019-01-12:
Despite being very unused to python and the gpio libraries used, I got the functionality running. Just had trouble with the python syntax building a press-handler that is only called when the held time is not reached. Somehow my pi is now dead after I was accidentally shorting some pins while trying to measure others with the multimeter :'(

Still struggeling with the full disconnection of the pi and external amp through gpio.

When My version runs finally, I will provide it as hack, like the other gpio-playout example. You can then consider whether you want to add it, or just keep it as "hack" for future developers to use parts of it.