MiczFlor/RPi-Jukebox-RFID #416: After first installation of Phoniebox + Spotify Edition sound stops working

chaskitb, 2018-12-29:


chaskitb, 2018-12-29:
Hello Gents
I'm experiencing a strange issue, after installing Phoniebox + Spotify Edition sound suddenly stops working (after entering first time the landing page of the Phoniebox). Even playing a file with aplay or so the speakers remain muted. I expect that it has to do something with the way the volume is controlled through the webpage. Any ideas how to solve this? 

My setup: 
Raspberry Pi Zero
Justboom Amp

MiczFlor, 2018-12-29:
Go to the terminal and type `alsamixer` 
For reasons, I don't know, the sound muted. And I think that, for reasons I also don't know. ... meaning: it might be something else, but try this first.
Thanks, m

chaskitb, 2018-12-29:
I tried it but without success, nothing in alsamixer is muted. As stated on the issue #359 I tried disabling the GPIO (sudo systemctl disable phoniebox-gpio-buttons), rebooted and that did the trick. I then activated the phoniebox-gpio-buttons, rebooted and the sound was gone again. My guess is that by what ever reason the gpio buttons always mute the output. My guess for all its worth either something in the webserver or in the gpio scripts. I'm though rather a beginner in programming and did not find anything (yet).

MiczFlor, 2018-12-30:
There could be a conflict between the sound card and the GPIO script, if you are using a mounted soundcard. Similar to this reported issue:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#using-phat-beat-dac-from-pimoroni
Because there are so many hardware options out there, it does not seem possible to make the one GPIO script that works for all.

chaskitb, 2018-12-31:
You were right, I had an overlap of the GPIO pin usage. I changed the pins used in the GPIO script to different pins that are not used in my setup and now the sound works like a charm. Thanks for your help! Guten Rutsch!

closed 2019-08-10 by MiczFlor