MiczFlor/RPi-Jukebox-RFID #417: Webgui - Mobile Phone - Timestamp out of view for playing song and playlist

SpookyFishes, 2018-12-29:
Hello,

on my mobile phone with 5" display the timestamp is out of the view for the actual playing title and for the playlist.

[image]

When there are folders with long names it looks like this:
[image]
Is it possible to cut the string at the end of the box and show the rest of the string in the next line?

Phoniebox SW
Version: 1.1.8
Edition: Plus edition (feat. Spotify integration)

gizmo21, 2018-12-30:
Plz check this too:

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/260

MiczFlor, 2019-02-05:
Should work now, Please check. Thanks for the feedback!

closed 2019-02-05 by MiczFlor

SpookyFishes, 2019-02-08:
The time which was out of range is fixed in 1.1.9-rc7 (I posted a pic here https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/260)

The problem with long names is still there

Thanks for spending time in fixing these issues.