MiczFlor/RPi-Jukebox-RFID #418: Add --yes apt install commands

lied, 2018-12-30:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7e12d45adfa566d80761af4adf3c87c5ecd21aa7/scripts/installscripts/stretch-install-spotify.sh#L459

This could be more "automated" when adding `--yes` to the command, as this automatically accepts the apt packages to be installed. (also relevant for the other install scripts as well)

MiczFlor, 2018-12-30:
Hi @lied 
this suggestion flatters me, because you seem to trust the install process :) I will keep this open and hope the day will come where I happily add this.

lied, 2018-12-30:
@MiczFlor hm, I don't get it, what bad could happen? Either it fails anyway after manually accepting the apt changes or it fails automatically :)

bbung, 2019-02-04:
+1 or add a --unattended option to the script so the user can choose to trust or not

MiczFlor, 2019-02-19:
I added this to the code.

closed 2019-02-19 by MiczFlor