MiczFlor/RPi-Jukebox-RFID #421: Adafruit PN532 breakout board - So close to a working setup, but cannot register my reader

imbaimbaimba, 2018-12-31:
I was initially inspired by this awesome project: http://shawnrk.github.io/songblocks/
and wanted to do a cool jukebox for my daughter.

I got my Raspberry Pi 3 running with an Adafruit PN532 RFID/NFC Breakout board following this guide: https://learn.adafruit.com/adafruit-nfc-rfid-on-raspberry-pi/overview and I am able to read my NFC tags perfectly now. Even has way better read distance than I assumed, so I am happy!

For testing, I used libnfc, that is included in the Adafruit guide. I go to the libnfc examples folder and run:
`sudo ./nfc-poll`

I get:
```
NFC reader: pn532_uart:/dev/ttyS0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
```

I then insert a tag and get:
```
ISO/IEC 14443A (106 kbps) target:
ATQA (SENS_RES): 00   04
UID (NFCID1):   XX   XX   XX   XX  (Different values for all my tags)
SAK (SEL_RES): 08
```

Then it stays there until I remove the tag and I get:
```
nfc_initiator_target_is_present: Target Released
Waiting for card removing...done.
```

I created the actual finished products to put on the wall for my daughter, and then realized I know absolutely nothing about coding. Therefore I had no idea where to start to actually get it to play anything.

I found your awesome project, and went through your guide: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch

Got everything running from SMB to webserver, and doing things manually it plays nicely out the USB speakers on my Pi! Awesome!

Now, I got no idea how to get my working reader implemented in my (other than reader) working Phoniebox. Before troubling you guys, I tried everything I could find.

I tried:
`python /RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py`
Which told me: "Please run RegisterDevice.py first"

`python /RPi-Jukebox-RFID/scripts/RegisterDevice.py`
Only gave me choices between my mouse and keyboard attached to the pi

I then copied the experimental scripts from here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special using:
`cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.experimental /home/pi/RPi-Jukebox-RFID/scripts/Reader.py`

I then tried RegisterDevice.py and daemon_rfid_reader.py again with the following result:
```
pi@raspberrypi:~ $ python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/home/pi/.local/lib/python2.7/site-packages/serial/__init__.py", line 27, in <module>
    from . import abc, model, marshal, errors, utilities, properties, meta, hooks, test, request  # noqa
  File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 44, in <module>
    from . import properties, meta, errors, hooks, abc
  File "/home/pi/.local/lib/python2.7/site-packages/serial/properties.py", line 34, in <module>
    from .abc.model import Model
ImportError: No module named abc.model
pi@raspberrypi:~ $ python /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 5, in <module>
    from Reader import Reader
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/home/pi/.local/lib/python2.7/site-packages/serial/__init__.py", line 27, in <module>
    from . import abc, model, marshal, errors, utilities, properties, meta, hooks, test, request  # noqa
  File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 44, in <module>
    from . import properties, meta, errors, hooks, abc
  File "/home/pi/.local/lib/python2.7/site-packages/serial/properties.py", line 34, in <module>
    from .abc.model import Model
ImportError: No module named abc.model
```

I would really appreciate it if you guys could help me out!

imbaimbaimba, 2019-01-02:
If someone can help me solve this, I'd be happy to pay a little donation to the helper! I really want this project done for my daughter, and just this one step missing :(

MiczFlor, 2019-01-03:
Hi @imbaimbaimba 
can you try the one line install again - and the different approaches you described in your last post. 
There were minor fixes since you last posted and they have to di with get_devices
The adafruit board, I don‘t know. Do you have a link?
All the best and good luck, micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

imbaimbaimba, 2019-01-03:
> Hi @imbaimbaimba
> can you try the one line install again - and the different approaches you described in your last post.
> There were minor fixes since you last posted and they have to di with get_devices
> The adafruit board, I don‘t know. Do you have a link?
> All the best and good luck, micz
> 
> Sent with [GitHawk](http://githawk.com)

Thank you for your reply, and thank you for providing such a great project! 

When I installed, I used the step-by-step guide, just to try and kind of understand what was going on. I just tried the one-line install, however the result is the same "ImportError: No module named abc.model"

About the Adafruit breakout board: https://www.adafruit.com/product/364
I have been recommended it from people using it for other things, as it should be extremely reliable and have a good read distance, that I need to read through a thick piece of oak.
The installation and testing guide on Adafruit describes it pretty well, and it worked great for testing: https://learn.adafruit.com/adafruit-nfc-rfid-on-raspberry-pi/overview

The project that inspired me, before finding Phoniebox, is using the same reader too:
Song Blocks: http://shawnrk.github.io/songblocks/ - https://github.com/shawnrk/songblocks 

Something similar going on here with a similar reader:
MPD-Box: https://mpd-box.readthedocs.io/en/latest/installation.html

I also found another project, where the author is using an Arduino Uno connected by USB.
MusicBox: https://maker.redgick.com/en/music-box - https://github.com/MusicBoxProject

imbaimbaimba, 2019-01-03:
I just took the plunge and created a new SD Card. Went straight to the one-line install, and now the error is different, but still missing modules. 

Still no resolution :(

`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python daemon_rfid_reader.py
Traceback (most recent call last):
  File "daemon_rfid_reader.py", line 5, in <module>
    from Reader import Reader
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/home/pi/.local/lib/python2.7/site-packages/serial/__init__.py", line 10, in <module>
    from . import utilities, abc, model, marshal, errors, properties, meta, hooks, test, request
  File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 11, in <module>
    from .utilities.compatibility import backport, BACKWARDS_COMPATIBILITY_IMPORTS
ImportError: No module named utilities.compatibility`

imbaimbaimba, 2019-01-03:
Still not having any success unfortunately, however I am now beyond the module problems (I think). I researched all day, and the closest I have been to anything is this testing sequence:

1) Tested first that it was working for polling tags with libnfc: Success.
2) Tried RPi-Jukebox-RFID and it gave me: ImportError: No module named abc.model
3) Tried to install serial with `pip install serial`, then I got: ImportError: No module named compatibility
4) Tested libnfc, still working.
5) Tried to uninstall serial with `pip uninstall serial`, libnfc is still working, and now there is no missing modules anymore apparently.
6) Tried RegisterDevice and saw my keyboard, mouse and MFRC522  RDM6300. 
MFRC522: IOError: [Errno 2] No such file or directory
RDM6300: Nothing happens till I CTRL+C - see code.
7) Checked libnfc, still working.

Any tips for where I could go from here?

Code:
`pi@raspberrypi:~ $ cd libnfc
pi@raspberrypi:~/libnfc $ cd libnfc-1.7.1
pi@raspberrypi:~/libnfc/libnfc-1.7.1 $ cd examples
pi@raspberrypi:~/libnfc/libnfc-1.7.1/examples $ sudo ./nfc-poll
/home/pi/libnfc/libnfc-1.7.1/examples/.libs/lt-nfc-poll uses libnfc 1.7.1
NFC reader: pn532_uart:/dev/ttyS0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 3c  61  10  35  
      SAK (SEL_RES): 08  
nfc_initiator_target_is_present: Target Released
Waiting for card removing...done.
pi@raspberrypi:~/libnfc/libnfc-1.7.1/examples $ cd
pi@raspberrypi:~ $ cd RPi-Jukebox-RFID
pi@raspberrypi:~/RPi-Jukebox-RFID $ cd scripts
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python daemon_rfid_reader.py
Traceback (most recent call last):
  File "daemon_rfid_reader.py", line 5, in <module>
    from Reader import Reader
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/home/pi/.local/lib/python2.7/site-packages/serial/__init__.py", line 27, in <module>
    from . import abc, model, marshal, errors, utilities, properties, meta, hooks, test, request  # noqa
  File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 44, in <module>
    from . import properties, meta, errors, hooks, abc
  File "/home/pi/.local/lib/python2.7/site-packages/serial/properties.py", line 34, in <module>
    from .abc.model import Model
ImportError: No module named abc.model
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ pip install serial
Collecting serial
  Downloading https://files.pythonhosted.org/packages/93/98/86b3ba751fd1dec0059e7fdc62495c62bbdd109bf470c9ec9f9c9e5d1759/serial-0.0.86-py2.py3-none-any.whl
Collecting pyyaml>=3.13 (from serial)
Collecting future>=0.17.1 (from serial)
Collecting iso8601>=0.1.12 (from serial)
  Using cached https://files.pythonhosted.org/packages/ef/57/7162609dab394d38bbc7077b7ba0a6f10fb09d8b7701ea56fa1edc0c4345/iso8601-0.1.12-py2.py3-none-any.whl
Installing collected packages: pyyaml, future, iso8601, serial
Successfully installed future-0.17.1 iso8601-0.1.12 pyyaml-3.13 serial-0.0.86
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python daemon_rfid_reader.py
Traceback (most recent call last):
  File "daemon_rfid_reader.py", line 5, in <module>
    from Reader import Reader
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/home/pi/.local/lib/python2.7/site-packages/serial/__init__.py", line 10, in <module>
    from . import utilities, abc, model, marshal, errors, properties, meta, hooks, test, request
  File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 11, in <module>
    from .utilities.compatibility import backport, BACKWARDS_COMPATIBILITY_IMPORTS
ImportError: No module named compatibility
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cd    
pi@raspberrypi:~ $ cd libnfc
pi@raspberrypi:~/libnfc $ cd libnfc-1.7.1
pi@raspberrypi:~/libnfc/libnfc-1.7.1 $ cd examples
pi@raspberrypi:~/libnfc/libnfc-1.7.1/examples $ sudo ./nfc-poll
/home/pi/libnfc/libnfc-1.7.1/examples/.libs/lt-nfc-poll uses libnfc 1.7.1
[... 76 lines cut ...]
/home/pi/RPi-Jukebox-RFID/scripts
^CTraceback (most recent call last):
  File "daemon_rfid_reader.py", line 16, in <module>
    cardid = reader.reader.readCard()
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 93, in readCard
    read_byte = self.rfid_serial.read()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 472, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ^C
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cd
pi@raspberrypi:~ $ cd libnfc
pi@raspberrypi:~/libnfc $ cd libnfc-1.7.1
pi@raspberrypi:~/libnfc/libnfc-1.7.1 $ cd examples
pi@raspberrypi:~/libnfc/libnfc-1.7.1/examples $ sudo ./nfc-poll
/home/pi/libnfc/libnfc-1.7.1/examples/.libs/lt-nfc-poll uses libnfc 1.7.1
NFC reader: pn532_uart:/dev/ttyS0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 3c  61  10  35  
      SAK (SEL_RES): 08  
nfc_initiator_target_is_present: Target Released
Waiting for card removing...done.
pi@raspberrypi:~/libnfc/libnfc-1.7.1/examples $`

xCite1986, 2019-01-03:
> File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 5, in <module>
>     from Reader import Reader
>   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module> [...]

got the same error and fix them with pyserial instead of serial

```
pip install pyserial
```

imbaimbaimba, 2019-01-03:
> > File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 5, in 
> > from Reader import Reader
> > File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in 
> > import serial
> > File "/home/pi/.local/lib/python2.7/site-packages/serial/**init**.py", line 27, in 
> > from . import abc, model, marshal, errors, utilities, properties, meta, hooks, test, request  # noqa
> > File "/home/pi/.local/lib/python2.7/site-packages/serial/model.py", line 44, in 
> > from . import properties, meta, errors, hooks, abc
> > File "/home/pi/.local/lib/python2.7/site-packages/serial/properties.py", line 34, in 
> > from .abc.model import Model
> > ImportError: No module named abc.model
> > ```
> > I would really appreciate it if you guys could help me out!
> > ```
> 
> got the same error and fix them with pyserial instead of serial
> 
> ```
> pip install pyserial
> sudo service phoniebox-rfid-reader restart
> ```

Actually, that part I fixed just by removing serial. 
`pip uninstall serial`

MiczFlor, 2019-01-04:
Hi @imbaimbaimba 
try to run these in the command line. This throws no errors for me. The `sudo pip install -r requirements.txt` way does throw errors. Strange I didn't notice before. Please try this and let me know if that works:
~~~
sudo pip install "evdev == 0.7.0"
sudo pip install --upgrade youtube_dl
sudo pip install git+git://github.com/lthiery/SPI-Py.git#egg=spi-py
sudo pip install pyserial
sudo pip install spidev
sudo pip install RPi.GPIO
sudo pip install pi-rc522
~~~

imbaimbaimba, 2019-01-04:
Hi @MiczFlor 
Even though that did install a lot of things that wasn't there before, it is still not working. Am I supposed to see my PN532 device when I run RegisterDevice.py ?

Rougie85, 2019-01-06:
I am getting the same error message while trying to register a RFID-Reader:

python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in <module>
    import serial
  File "/usr/local/lib/python2.7/dist-packages/serial/__init__.py", line 10, in <module>
    from . import utilities, abc, model, marshal, errors, properties, meta, hooks, test, request
  File "/usr/local/lib/python2.7/dist-packages/serial/model.py", line 11, in <module>
    from .utilities.compatibility import backport, BACKWARDS_COMPATIBILITY_IMPORTS
ImportError: No module named utilities.compatibility

Installed it today by the help of the "one line install command" (Spotify Edition).
Any ideas?

Thanks in advance.

imbaimbaimba, 2019-01-06:
> I am getting the same error message while trying to register a RFID-Reader:
> 
> python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
> Traceback (most recent call last):
> File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 4, in 
> from Reader import get_devices
> File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 10, in 
> import serial
> File "/usr/local/lib/python2.7/dist-packages/serial/**init**.py", line 10, in 
> from . import utilities, abc, model, marshal, errors, properties, meta, hooks, test, request
> File "/usr/local/lib/python2.7/dist-packages/serial/model.py", line 11, in 
> from .utilities.compatibility import backport, BACKWARDS_COMPATIBILITY_IMPORTS
> ImportError: No module named utilities.compatibility
> 
> Installed it today by the help of the "one line install command" (Spotify Edition).
> Any ideas?
> 
> Thanks in advance.

If you do this, I think you will get past that error:
`pip uninstall serial`

closed 2020-01-18 by MiczFlor