MiczFlor/RPi-Jukebox-RFID #423: New Feature: German language file

szeberli, 2019-01-03:
Ich habe die Sprach Datei übersezt. und einen Patch gemacht. 

**Patch:**
[RPi-Jukebox-RFID-Language.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2724649/RPi-Jukebox-RFID-Language.zip)

**Sprache DE Datei:**
[lang-de.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2724371/lang-de.zip)

**config.php ergänzen:**
`$conf['language']   = 'de'; // Set Language`

Viel Freude :)

wuchdi, 2019-01-17:
Erst einmal viel Dank dafür. 
Hast du einen Hinweis, wie man den Patch installiert?

Der Speicherort für lang-de.php und die Ergänzung der config.php war kein Problem.

MiczFlor, 2019-01-20:
@RogoTe teil doch schon mal was du hast. Bis jetzt ( Stand Sonntag 08:56) kam ich noch gar nicht an den Rechner. Das geht glaube ich vielen Phonieboxlern so. Kinder... :)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nubbler, 2019-01-21:
Wäre auch für einen Hinweis, wie man den Patch installiert dankbar.  Der Speicherort für lang-de.php und die Ergänzung der config.php war auch für mich kein Problem, hier hören meine Kenntnisse dann aber auch schon auf :-).

MiczFlor, 2019-01-22:
Hallo,
ich habe das Sprachfile von @szeberli jetzt mit master gemerged: #451 
Ein "Howto" wie man das File erweitert / bearbeitet kommt später.
Achtung: man braucht jetzt nicht mehr die config.php zu patchen. Die Sprache kann im WebUI unter "Settings" ausgewählt werden.
Oder, wie es dann heisst: "Einstellungen" :)

closed 2019-01-22 by MiczFlor

iroQuai, 2019-04-30:
So, i believe it now is possible to add other languages? I'd like to add Dutch too. 

(Sorry, my german isnt good enough to see if the answer to my question is already answered...)

MiczFlor, 2019-05-04:
Hi @iroQuai 
I have to write this up in the wiki. Sorry that I didn't round to doing this. Essentially what you need to do:
* make a copy of the file `lang/lang-en-UK.php` which you can [find here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/lang/lang-en-UK.php)
* save it in the same folder and call it `lang-nl-NL.php`
* translate the strings and save it
* it should come up in the web app interface under settings
Please commit your translation or post it here, so I can commit it.
Thanks for adding to the code base!!!
micz

iroQuai, 2019-05-10:
Thanks, that was really helpful. In my opinion you can just copy-paste this
text into the wiki.

I created the dutch language file and submitted a pull request!