MiczFlor/RPi-Jukebox-RFID #424: Startupsound is triggered to early (spotify edition)

petschni, 2019-01-03:
In the spotify edition the startup sound is triggered too early. I seems like the sound is triggered once Phoniebox is started but that does not mean that the "spotify service" is good to go. I usually have to wait for another ~10 seconds before rfid cards with spotify songs will work. 

Is that a bug in my setup or happening to other people as well?

c0un7-z3r0, 2019-01-09:
I'm having the same issue. But MP3s that are directly on the Pi work straight away after the sound.

bbadaboom, 2019-04-07:
I have the same issue but an even longer lag for the google play music service (20-30 secs), would be great if anyone with a solution can help us out (even a second sound implying that it is now possible to stream would help significantly).

domu83, 2019-04-08:
similar on my side... I even do not get the startup sound through the USB soundcard as it seems not ready when the sound is playing. When I shutdown using the WebUI I hear the shutdown sound, so the playout seems to work but just earlier.
For Spotify I thought issue might be unstable WIFI connection. Would it be possible to have a sound as soon as a Internet Connection is established?

petschni, 2019-04-16:
@MiczFlor I think the issues are a bit different, or? The issue is described in the other thread is about the sound not playing at all. The issue I described is more like that the sound in general seems to work but the "i am ready" sound does not take the startup time of the "spotify plugin" into account.