MiczFlor/RPi-Jukebox-RFID #425: Different Buttons reactions

splitti, 2019-01-05:
Hello,

i just want to make a new functionality to next and prev buttons for long pressing. Is there a solution for that?
In my case, i want a normal shot press for next or prev, on long press (2 secs) i want to change the contrast of my Oled Display (https://github.com/splitti/oled_phoniebox) up or down.

Hope it is possible, because i want as less buttons as possible.

Wallnusskaese, 2019-01-06:
You can do this already. 

I am running 9 GPIO buttons which are linked to PseudoRFID-Codes (0-9). 
Short press play/pauses.
Long press restarts the playlist.
Double click is connected to playernext.

For double click the script just checks whether two presses (of the same button) are within a time period.
For distinguishing between short and long press, the time between button press and release is being measured. 

Here is my gpio-buttons.py. It's quite messy as I didn't worked with python before, but I hope you get the logics. 
https://github.com/Wallnusskaese/RPi-Jukebox-RFID/blob/master/scripts/gpio-buttons_OLD.py

splitti, 2019-01-08:
Thanks, found it by my own, but your code looks simply clean, nice! Thats what i want, so i can go on with next improvments of the display ;-)

closed 2019-01-08 by splitti

splitti, 2019-01-08:
Here is my code, please notice, i am also working on my project...
https://github.com/splitti/oled_phoniebox/blob/master/scripts/gpio-buttons/gpio-buttons.py