MiczFlor/RPi-Jukebox-RFID #428: spend some coffeecups

ddbkva, 2019-01-07:
hey
i would like to spend some coffeecups like tinyCoffee or give for wordpress.
I just finished the third box and would like to say thank you
(its not my last box)

MiczFlor, 2019-01-08:
First: post your pics from the box :) that‘s also motivating!
Secondly: I will look into this. It might help me to really allocate time for maintenance and support on my end. 
Thanks for the positive energy. All the best, micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MiczFlor, 2019-01-08:
Hi @ddbkva 
you asked for it, I never tried it and just set it up:
https://www.buymeacoffee.com/MiczFlor
You might be the first with three Phonieboxes AND the first to buy me a coffee :)
Feel free to promote this option. As I said above, it might actually make sense for everyone, including me, if I can dedicate time to Phoniebox.

ddbkva, 2019-01-08:
cheers

WIMS2, 2019-05-18:
Wie wäre es mit PayPal.Me? Würde gerne einen Kaffee kaufen für die tolle Arbeit 👍, aber ich habe nur die Kreditkarten-Option gefunden.
Fotos der Box finden sich im RPI-Forum.

MiczFlor, 2019-05-19:
Hi @WIMS2 
thank you for the offer :) You can PayPal me privately at micz.flor@web.de

SirUli, 2019-12-07:
@MiczFlor you might want to take advantage of Github sponsors: https://help.github.com/en/github/supporting-the-open-source-community-with-github-sponsors

Piehti, 2020-02-20:
Just spent some Coffees via PayPal. Enjoy mate. 
Thanks for the code & documentation.

Peter, Jessica & Linett.