MiczFlor/RPi-Jukebox-RFID #429: Change Countdown to stop playback instead of shutdown

Joebinator, 2019-01-08:
Hi there!

Is there anyhow a solution where i can stop playback instead of shutting down the entire phoniebox?

My daughter likes to listen to music while sleeping time and i she also likes do hear music when she wakes up. But if the box is turned off she sometimes don't get it to restart the box entire.

MiczFlor, 2019-01-08:
Good idea. I will put this into my todo list. All the best, micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Klickmichel, 2019-01-11:
I think it ist possible to config a RFID trigger to stop playback after 5, 15, 30 or 60 minutes.
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/settings/rfid_trigger_play.conf.sample

Joebinator, 2019-01-28:
hey, sorry for the late response.

the rfid trigger solution works fine so far for me :)

closed 2019-01-28 by MiczFlor