MiczFlor/RPi-Jukebox-RFID #430: rfid_trigger_play.sh

StL2412, 2019-01-08:
Hi, how can i open the file: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh?
Everytime i tried the following message showed up: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 280: [: too many arguments

And i have a Problem with this file: $ nano /home/pi/RPi-Jukebox-RFID/settings/rfid_trigger_play.conf
after i hit enter this message showed up:
-bash: $: command not found

can someone help me please

MiczFlor, 2019-01-08:
Hi @StL2412 
did you catch that error? I haven't seen the line 280 issue and wonder what it could be.

StL2412, 2019-01-09:
Hi MiczFlor,

Ich habe vergessen das Nano vor die Eingabe zu schreiben bzw. beim zweiten das $-Zeichen nicht entfernt. Daher kamen die Fehlermeldungen. Nachdem ich es korrekt ausgeführt habe, hat auch alles geklappt.

Mit freundlichen Grüßen 
StL2412