MiczFlor/RPi-Jukebox-RFID #434: Spotfy edition one line install files partly

cubertt, 2019-01-09:
I tried to install the spotify edition on a raspberry zero w with the script ( stretch Desktop (full  2018-10-09-raspbian-stretch) and iris is not working, i think because of this error:

`pylast requires Python '>=3.5' but the running Python is 2.7.13`

cant find a solution, any hints?

cubertt, 2019-01-10:
This seems to be an issue with iris, so i looked in the github issue page of iris and found:

pip install 'pylast<3.0'

i have than managed to install iris via git clone ... and manually running setup.py (and updating some suggested things, because the error still occurred.. ) than finally it installed fine...

MiczFlor, 2019-01-10:
Hi @cubertt 
thanks for sharing your knowledge. Can you be a bit more specific in what made it work - if others find this here? I know, once it works, it works and it is difficult to retrace the steps. But you might remember a few more hints :)
Thanks, micz

cubertt, 2019-01-11:
I tried it again, but 

`$ pip install 'pylast<3.0'`

does not help.

Could install Iris via python3 runs without errors

then installed phonibox script --> iris is not working (phyton2 vs python3?)

So, no i cant give more information, i have a working version but dont know why 

Can someone confirm the error?

Berdsen, 2019-01-11:
I had the same error and the problem and solution is described in pull request #436

closed 2020-01-18 by MiczFlor