MiczFlor/RPi-Jukebox-RFID #435: Install on Armbian 5.9 (Xunlong Orange Pi Zero)

dnmtch, 2019-01-10:
tl:dr; I was unfortunately not able to install RPi-Jukebox-RFID on an Xunlong Orange Pi Zero / Armbian 5.9. The manual installation works, but there are probably problems with scripts. (RegisterDevice.py not working)

Lieber MiczFlor, zuerst einmal ein riesiges Dankeschön für deine Arbeit.
(wenn ich störe, Issue einfach schliessen)

Ich habe eine manuelle Installation auf einem Xunlong Orange Pi Zero mit Armbian 5.9 versucht. Leider hat das (für meinen Wissensstand) nicht so richtig funktioniert. Ich vermute das fehlende "home/pi" Directory und gewisse Raspberry-Eigenschaften funktionieren nicht. 

**Hat jemand eine Installation auf einem Orange Pi Zero (Armbian) hingekriegt und kann helfen?**
(gibt es Hilfe für nicht-Raspberry Installationen?)

MiczFlor, 2019-01-10:
Hi @dnmtch 
I answer in English, for others to contribute (assuming that English would still give you the broadest support group :)
I have no experience with the hardware you mention - and agree with you that this might well also go into hardware differences. Especially the `evdev` package, which is central, does grab deep into the hardware of the RPi. So my short and only anser is: sorry, I hope somebody else can help you...
And I want to add: if you figure this out, please document it in the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki

dnmtch, 2019-01-12:
Ok, up an Running!
(RFID Reader not working jet)

Setup:
- Xunlong Orange Pi Zero (with USB/Audio-Board Extension Borad)
- Armbian_5.69_Orangepizero_Debian_stretch_next_4.19.13

Install:
- apt-get update
- apt-get upgrade
- armbian-config (enable analog sound)
- adduser pi 
- usermod -aG sudo pi (make pi user root)
- su - pi (change to user pi)
- sudo apt-get install python-pip
- sudo pip install --upgrade pip setuptools wheel  (fix for python setup.py egg_info" failed with error)
- install manually -> [Install required packages and the Phoniebox code](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#install-required-packages-and-the-phoniebox-code)
- run Setup Script (Phoniebox Classic) 

Sound Configuration:
http://esp8266-server.de/OrangePi/#Webradio 
/etc/mpd.conf
```
type	"alsa"
name	"My ALSA Device"
device	"hw:0,0"	
mixer_type "software" 
```

MiczFlor, 2019-01-13:
Hi @dnmtch 
Great news. If you find the time, could you create a page in the wiki for this hardware setup? All the best, micz  

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-01-18 by MiczFlor

dnmtch, 2020-05-28:
Update:

apt-get update
apt-get upgrade

**Audio**
Follow this instructions: https://forum.armbian.com/topic/3446-orange-pi-zero-no-audio-devices-available-cannot-play-audio/?tab=comments#comment-86786
- armbian-config (enable analog sound -> armbian-config -> System -> Hardware: select analog-audio with the space-key)
- reboot and check aplay -l shows, it shoud be

> 
> aplay -l shows
> 
> **** List of PLAYBACK Hardware Devices ****
> card 0: Codec [H3 Audio Codec], device 0: CDC PCM Codec-0 []
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0

3.  create file .asoundrc in home directory
> nano ~/.asoundrc

put this in

> pcm.!default{
>     type hw
>     card 0
> }
> 
> ctl.!default{
>     type hw
>     card 0
> }

type alsamixer 
[image]
test Speaker
speaker-test -twav -c2 -Dhw:0 

adduser pi
usermod -aG sudo pi (make pi user root)
su - pi (change to user pi)

install script **A. Default install script for Buster (recommended)**
(once as User "pi", once as root 

[image]