MiczFlor/RPi-Jukebox-RFID #437: Tweak recorder functionality by adding "playback" functionality

comdoxx, 2019-01-12:
We have the "Brio Sprechender Bahnhof" which has similar functionality.
It has a red button that records whenever pressed and a green button that plays the last(!) recording.
I'd like the jukebox to do the same, so in gpio-buttons.py there hast to be something like:
```
# Todo: How to specify (max) recording time?
def def_startrecord():
  check_call("./scripts/playout_controls.sh -c startrecord 10", shell=True)
def def_stoprecord():
  check_call("./scripts/playout_controls.sh -c stoprecord", shell=True)
[...]
reco = Button(21,pull_up=True)
[...]
reco.when_pressed = def_startrecord
reco.when_released = def_stoprecord
```
Now, the only thing missing is the playback of the last recording. My suggestion is to add another command such as playback to the playout_controls that triggers the playback of the latest recording.

Bonus: 
- Put the recording device configurable outside the playout_controls.sh
- mkdir if not exists ("Recordings")
- make configurable whether "normal playback" stops or continues
- maybe (then) use mpc instead of aplay (which would be the counterpart of arecord)
- if aplay is used: "killall aplay" before calling arecord.
- using aplay / arecord while mpc playback could lead to either cool sound overlays (although only one sound possible) to mpc playbacks or to your kids recording sing-alongs... ;-)

_Originally posted by @comdoxx in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/177#issuecomment-452483775_

comdoxx, 2019-01-12:
Put this into Pull Request 440:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/440

MiczFlor, 2019-01-13:
Hi @comdoxx 
I merged your code with the `develop` branch. And then made some minor naming tweaks. Your "playback" feature is great and in tune with the BRIO funcionality.
I renamed all recording playout commands like this:
~~~
# recordstart
# recordstop
# recordplaylatest
~~~
To make this easier to read, consistent and easier to expand in the namespace / feature of "recording".
Thanks for your contribution :)

comdoxx, 2019-01-13:
I like the naming, but I didn't want to change naming style, therefore let it as it was.
My "#Todo" from the thread opening post is now answered: It would be the -v option. Should / Could also be used in `misc/sampleconfigs/gpio-buttons.py.sample` as the max time.
Naming now is inconsistant. `scripts/rfid_trigger_play.sh` still uses `STARTRECORD` and `STOPRECORD`. But changing this could harm existing installations. I'd nevertheless prefer this to be changed.
You probably may want to add
```
        $RECORDPLAYBACKLATEST)
            $PATHDATA/playout_controls.sh -c=recordplaybacklatest
            ;;
``` 
to `scripts/rfid_trigger_play.sh` at line 236 to have the ability to swipe such a card?

MiczFlor, 2019-01-22:
Hi @comdoxx 
I went the soft way :)
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/437

closed 2019-01-22 by MiczFlor