MiczFlor/RPi-Jukebox-RFID #442: Fix inside: youtube-dl not working on after some time (or on older release)

gizmo21, 2019-01-13:
I'm still on v.1.0.0 (of course classic ;) )  with my box as this is the most stable/fast version for me and nearly has all the features I need. But youtube downloads failed some weeks ago and I now had time to check and a simple _sudo apt-get upgrade_ didn't do the job as youtube-dl came not as apt package but as python pip. I guess especially the youtube-dl service needs some regular updating as yt will change it's site/protocol from time to time, so even on the lastest release-version this could happen if there is no new version is out for some weeks.

My problematic version was:
_Found existing installation: youtube-dl 2018.9.8_
Which did not display any errors, but just did not download anything to the folder.
Manually check the error by using:
_youtube-dl -f bestaudio --extract-audio --audio-format mp3 https://www.youtube.com/watch?v=7GI0VdPehQI_

> Error: _RegexNotFoundError: Unable to extract Initial JS player signature function name; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output._

Was fixed with:
_Successfully installed youtube-dl-2019.1.10_

I would paste the following into the wiki-FAQs if it is OK:

----------------------------

### Fix broken youtube-dl while staying on same RPi-Jukebox-RFID releaseversion 

Youtube download could fail after some time (due to site/protocol... changes). There could be no error displayed in webfrontend and the log also does not show an error. To check if the service has a problem just manually run:
_youtube-dl -f bestaudio --extract-audio --audio-format mp3 https://www.youtube.com/watch?v=7GI0VdPehQI_ 
If it says update youtube-dl and if you don't want a full phoniebox reinstall or a regular RPi-Jukebox-RFID update to the newest releaseversion here is the way to update youtube-dl only:
_sudo pip install --upgrade youtube_dl_
This will only upgrade this service and will leave the rest of your install untouched.

----------------------------

gizmo21, 2019-01-14:
As no one answered negativly, I put it here: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#fix-broken-youtube-dl-while-staying-on-same-rpi-jukebox-rfid-releaseversion

MiczFlor, 2019-01-15:
Hi @gizmo21 
Here comes my positive answer: I had the same issue, solved it like you and failed to add it to the wiki :) thanks for the contribution!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

gizmo21, 2019-01-18:
Had a problem again with a video and updated to 2019-01-17. I wrote a script to do this update from webfrontend, for all that use these scripts: http://<yourip>/userScripts.php:

/home/pi/RPi-Jukebox-RFID/script/userscripts/updateyoutube-dl.sh   (Filerights:755)

```
#!/bin/bash

# From time to time youtube changes it's page/protcols and youtube-dl does not work anymore
# This script just updates the python pip package 'youtube_dl'
# no options in this script
echo '-------updateyoutube-dl.sh -------' >> /home/pi/RPi-Jukebox-RFID/scripts/userscripts/userscript.log
sudo pip install --upgrade youtube_dl >> /home/pi/RPi-Jukebox-RFID/scripts/userscripts/userscript.log
```

closed 2019-01-18 by gizmo21