MiczFlor/RPi-Jukebox-RFID #444: New to GPIO and RaspPi - how to implement an on/off button?

derOger, 2019-01-15:
Hello, can you help me with that? I want to build a Jukebox myself and think about the hardware needed.
I would prefer to use a small screen for informations or at least some 8-segment-displays for track number if possible.
Also since I want to use a powerbank I wonder how to implement a power button. For shutdown a simple halt would be enough but I think since there are speakers and other involved maybe this isn't enough. So there are cables which cut of the power via power switch so maybe using this between bank and RaspPi would be an option - but would that harm the OS when used often since they just cut of the power to the Pi while running.
How do you solved this?
Regards and sorry for posting on issues - I don't know where to ask elsewhere ...

derOger, 2019-01-15:
OK, as far as I read - sorry, I was to fast to post, I just connect Pin 5 and 6 with a button for shutdown?

derOger, 2019-01-15:
Thank you. Can you recommend a ATXBoard which is easy available in germany?
Regards

MiczFlor, 2019-01-15:
Hi @splitti 
Wouldn‘t it be a great contribution to create a pull request with your code for the oled to the develop branch?
And I really like your blog post(s) by the way!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MiczFlor, 2019-01-15:
@derOger
I can‘t find the link, bit there was a recommended hardware on/off shim in another thread - sent from the UK via ebay. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

splitti, 2019-01-15:
Look in my Blog Post, there are different solution for the ATX Boards. But i also wait for YKRUD3 and OnOff Shim because of the prices...
Micz, Yeah of course, but i am really New to Github, maybe you can help me

splitti, 2019-01-15:
https://www.ebay.de/itm/Pimoroni-PIM269-OnOff-SHIM-for-Safe-Shut-Down-for-Raspberry-Pi/123190210907?hash=item1caeb5715b:g:er0AAOSwRQlbImvx

derOger, 2019-01-17:
Thank you, So I connect the Shim with  the button and with the power input of the PI? The power supply is my biggest problem at the moment because I want it be easy to turn of and on. Especially when using the power bank.

splitti, 2019-01-17:
Yeah, the OnOff Shim goes on different GPIOs, because you have to communicate with the OS. It is not just pressing on or off, maybe you want to idle Shutdown? And the Controller Shutdown means, that the OS will halt before the power cut. That's why you need different GPIOs.

puked79, 2019-01-17:
I used onoffshim and supercaps (and a small resistor - without one my powerbank would detect a short immediately when switched on)

Powerbank------switch(4 pin on/off switch - 1 for the shim, the other one to cut the power)----supercaps---raspi

The (my) Problem with the shim was - if I only power down the pi with the shim, my powerbank will go into standby after some time, and that causes some unwanted behaviour. 

wired like this, the pi has enough time to power down after it gets the signal from the shim due to the supercaps, And it will turn on again.

splitti, 2019-01-17:
@puked79 
which Powerbank?

splitti, 2019-01-18:
Ok, ty for the Info. I just search for other Powerbanks. So, i am interested, if the problem is also there with my Bank an the Shim.

puked79, 2019-01-18:
Try it, I guess :/
And give feedback when you did :) (I am confident, my first phoniebox was not my last...)

iwishitwassummer, 2019-01-20:
> I used onoffshim and supercaps (and a small resistor - without one my powerbank would detect a short immediately when switched on)
>
> Powerbank------switch(4 pin on/off switch - 1 for the shim, the [...]

Hi @puked79, could you add a link to the supercaps please? Do you have a photo of the wiring by chance? Many thanks

puked79, 2019-01-22:
@iwishitwassummer no - that won`t work. The raspi would not shut down - or it shuts down and restarts (I tried that, but don`t remeber what exactly was the problem).

the shim triggers a script that shuts down the raspi and afterwards it cuts the power. 
You have two options - 

first:
the shim has to be plugged in all the time (5 Volts), else it won`t work.
This will not work with most powerbanks - they go into standby when no electrical consumer (is this the correct term?!) draws power. thus the shim won`t work. That was a big letdown for me, when I found out...
second:
that`s what I did - I used a 4 pin on/off switch one circuit cuts the power at cable between powerbank and raspi - directly behind the powerbank, the other one triggers the shim (You have to change a line in the script to make this work).

like this: Powerbank---switch circuit 1-------small resistor---supercaps------XL6009 DC-DC boost module-----shim at switch circuit 2------raspi

I connected the shim via jumper cable - you only need 4 pins connected - you`ll find the used pins on their homepage in the comments (or via google i guess)

puked79, 2019-01-22:
supercaps have to be in row!

SpookyFishes, 2019-01-22:
I am using ATXRaspi instead of on / off shim. Here are my experience with this product: 
https://forum-raspberrypi.de/forum/thread/41534-powerbank-versorgt-pi-und-shutdown-ueber-atxraspi/

I had the same problem that my powerbank went in standby and I wasn't able to wake the powerbank by using the ATXRaspi.
My soloutin is this little boad: https://forum-raspberrypi.de/forum/thread/28272-powerbank-zur-versorgung-kleiner-lasten-nutzen-grundlast-fuer-powerbank/
It is plugged in the powerbank and every 5s the little board consumes 200mA. With the hack the powerbank doen't go to sleep an my ATXRaspi is working fine.

derOger, 2019-01-23:
Hello, do you experience a real power drop by that? And is there a way to get such a stick? I don't have the equipment to build it myself (or is there a blueprint for - ehr - Lochplatinen (?!?)?

Gelegenheitsbastler, 2019-01-23:
Hi,
there is no blueprint available. Ask the user "Neueinsteiger". He will build one for you.

SpookyFishes, 2019-01-23:
@derOger My Powerbottun is connected to the ATXRaspi board. When I press the button longer then 2s the ATXRaspi will shutdown the Pi and if this is finised the ATXRaspi board disconnect the power supply to the pi.
The stick is just for keeping the powerbank alive.

derOger, 2019-01-24:
@SpookyFishes Thank you. I fear that I will run into the same problems (Powerbank goes to sleep) so I want to bypass this problem early on. So I search for a prooved solution I can use.