MiczFlor/RPi-Jukebox-RFID #448: Neuftech RFID-Reader - how to deactivate beeper?

derOger, 2019-01-17:
Hello,
since this is one of the recommended ones maybe someone had this issue (?). Can I mute the Neuftech RFID Reader? Now the reader beeps loudly when it is turned on and when I scan a card. Can I deactivate this beep? I have found some Java code for that but no other option.
Thank you alot.

tobiasnix, 2019-01-17:
Its not possible to deactivate. But you can easily remove the hardware component. See also #223

derOger, 2019-01-17:
I thought about opening the case and removing the beeper. Also to maybe save some space in the case later on ...
But the most waste of space are the cables I think.

MiczFlor, 2019-01-22:
I close this, I think it's solved. Silence in the forum == beeper ripped off the board :)

closed 2019-01-22 by MiczFlor