MiczFlor/RPi-Jukebox-RFID #449: Plans for changing from Wifi-Client to AP and back?

splitti, 2019-01-18:
Sometime you have to add a new Wifi SSID (holidays, Hotel, Wifi Connect for Spotify). But if you are away from your current Wifi, this gets a little problematic. So i've read about the method to set the phoniebox to an Access Point.
Are there any plans to change from Client to AP and back by Rfid Card?
Maybe you have also other ideas for that problematic?!

gizmo21, 2019-01-18:
There are several topic about that already in issues and wiki:
e.g.:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/297
or
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#user-content-addmultiplewifissids

The most simple way is to add your mobile hotspot once when you are at home via:
http://<yourraspi_IP or raspi_hostname>/userScripts.php an select addhotspot.sh with the right options.

Everytime you are away from home:
Then when your at a hotel just open your mobile hotspot to get connection to your phoniebox and then again add your hotel SSID via:
http://<yourraspi_IP or raspi_hostname>/userScripts.php
Then restart phoniebox and disable you mobile hotspot in your phone and you should be running.

MiczFlor, 2019-02-19:
Hi @gizmo21 
thanks for jumping in here. @splitti I never got the AP to work. The main issue I had is that it only works with "some" hardware, not all of it. This would mean that the installation would get more complicated. And some would try anyway, even though they don't have matching hardware - and then need to wipe the card, because the "uninstall" of AP is not trivial.
@gizmo21 possibly we should add your script to the web ui more visibly...?

Yordan1976, 2019-02-19:
And please make addhotspot.sh executable :) I didn't find out how to make a pull request for just a change in chmod settings yet.

MiczFlor, 2019-02-19:
@Yordan1976 - I was actually thinking about deleting it, because it does not work... I don't want people to ruin their setup with half baked ideas that I kept in the core, simply because I never expected so many people to dig through my "work in progress" ;)

Yordan1976, 2019-02-19:
Well, it works when it is executable :)
btw: welcome back!

gizmo21, 2019-02-19:
@MiczFlor & @Yordan1976 : I guess you two are talking at cross purposes, cause this thread is "AP" and "Addhotspot.sh" together. Perhaps this thread should close and continue at each original threads.

@MiczFlor you are way better implementing a gui for the sh or integrate even the sh better as I just recycled your html of version 1.0 for the UserScripts.php 
But then there is still no management for the hotspots, it's still a fire and forget thing that needs manual config editing if change or delete is needed.  That said for many people it still seems to make the deal when going to another location, so no problem for me if you move that the elsewhere in the gui. But would like to see userScripts.php still in the code there as it already has some additions on my side like youtube-dl-update.sh or other small scripts to make life easier without always needing putty.

closed 2020-01-18 by MiczFlor