MiczFlor/RPi-Jukebox-RFID #450: Rotary encoder for forward/backward seek?

iwishitwassummer, 2019-01-19:
Hi all,

did anyone tried to do use a rotary encoder for seeking?
My kid is using the box foremost to listen to audiobooks and wants to seek to the favorite parts of the audiobook. In combination with @splitti OLED it might be be possible to visualize the "progress".
Any advice is welcome, I'm a newbie to GPIO.
TX

iwishitwassummer, 2019-01-20:
Hi @bofelix 
it should be possible to use the [HTTM button](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5#issuecomment-455779732) for this, what do you think?
The time between the high and lows of the button should be identifiable which could be used to seek a track, right?

iwishitwassummer, 2019-01-21:
Hi,
Ich meine vorspulen (nicht den Track wechseln). Wenn der Button im hold Mode ist, könnte man doch die Zeit zwischen den 2 Signalen messen und daraus den „Vorsprung“ in einem Track nutzen, richtig? Klar, ich muss  noch Code schreiben und ich weiß nicht ob eine Berechnung schnell genug geht.
Danke dir simisch

bofelix, 2019-01-21:
Hallo 

Deutsch ist einfacher.
Ich bin leider kein Python Profi.
Aber es gibt die Möglichkeit Aktionen zu definieren für gehaltene Knöpfe. So ist es beim Shutdown. Da steht im Code shut when held. Du kannst dann auch bestimmen wann ein Knopf als gehalten erkannt wird. So habe ich mir den Playknopf umprogrammiert zum Herunterfahren nach 2 Sekunden.
Ob der MPD aber spulen als Kommando kennt weiß ich nicht.

Gruß Stefan

iwishitwassummer, 2019-01-21:
Danke Stefan. Dann konntest du für Lautstärke und Skip konntest du den vorhanden Code nutzen? 
Vorspulen geht, ist schon im Standardpaket drin. Im Moment mach ich das per Karte. Aber das ist sehr nervend und dauert. Ist auf 15s eingestellt und man muss immer die Karte auf den Reader halten für jeder weitere Sequenz. Selbst bei Erhöhung der Vorspulzeit wird das in einem Hörbuch nervend, und dann noch dieser Reader Beep :0

bofelix, 2019-01-22:
Hallo.

Ja das Pythonscript gpio-buttons.py hat sofort funktioniert mit "pullup false".

Gruß Stefan

iwishitwassummer, 2019-01-22:
Super danke dir. Buttons sind bestellt, ich wage den Versuch.

closed 2019-01-22 by iwishitwassummer

bofelix, 2019-01-24:
> @bofelix Habe die Taster bekommen, für Play/Pause mal angeschlossen.
> Die gute Nachricht, es geht. Der Taster is im Latch Modus. Laut
> Datenblatt: >

Hallo.

Latch ist halten....

Ich habe bei allen Schaltern die Lötbrücke gesetzt.

Gruß 
Stefan

iwishitwassummer, 2019-01-25:
@bofelix Alles klar danke dir