MiczFlor/RPi-Jukebox-RFID #454: Problems with the appearance of the web app

christianvoss90, 2019-01-22:
I have a problem with the web app. The labels for most fields are not displayed. The whole with different devices (Mac Book, Android Smatphone) and with different browsers. I have a screenshot here: 
[image]

MiczFlor, 2019-01-23:
That doesn‘t look right :) can you try „git pull“ again in the terminal?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mannifredl, 2019-01-23:
Same issue here. I did a git pull but everything is up to date already.
i used the one line installer script for the classic phoniebox if that helsp anything
thanks for your great work  :)

MiczFlor, 2019-01-23:
Hi @mannifredl @christianvoss90 
I will fix it tonight. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mannifredl, 2019-01-24:
Wohooo, great work! Thanks a lot
My daughter loves you ;)