MiczFlor/RPi-Jukebox-RFID #456: capacitive touch sensor ttp223

habersatt, 2019-01-24:
Hello!
Instead of push buttons, I have installed capacitive touch sensors type ttp223. Everything works great except for volume up and down and mute.
My question: How do I adjust the gpio-button.py so I can adjust the volume. Maybe someone has already gained experience. 
Thanks in advance.
Clemens

comdoxx, 2019-01-25:
Hi habersatt,

If everything is wired correctly (VCC, GND and GPIO) and you didn't bridge either A or B or the ttp223's board, it should behave just as a normal button, and adjust volume in both ways: Single touch and holding. Make sure to connect the correct GPIO you take from the `gpio-button.py` to the sensor. Don't mix up pin number and GPIO number.

If that's done and you still experience trouble, feel free to post (the excerpt of) your gpio-button.py and maybe a photo of your connection.

comdoxx, 2019-01-26:
> Hi comdoxx,
> Unfortunately, the sensor does not behave as it should. After a touch, the
> volume decreases continuously to 0. If I set the volume to 100, this
> process repeats itself. An ordinary push button makes everything right on
> the same pin.
> Since I used the sound over a pifiberry dac +, I chose the flexible
> pinout. Attached now so my gpi-button.py and photos of the wiring.
> Thanks and have a nice weekend
> Clemens
>
> #!/usr/bin/python3
> from gpiozero import Button
> from signal import pause
> from subprocess import check_call
>
> def def_shutdown():
> check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)
>
> def def_volU():
> check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
>
> def def_volD():
> check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)
>
> def def_vol0():
> [... 36 lines cut ...]
> <https://user-images.githubusercontent.com/47001677/51786442-ae866400-2164-11e9-8189-d11264a85a1e.jpg>
> [image: img_4378]
> <https://user-images.githubusercontent.com/47001677/51786443-ae866400-2164-11e9-82fc-cb6391fe3eec.jpg>
> [image: img_4377]
> <https://user-images.githubusercontent.com/47001677/51786441-ae866400-2164-11e9-8691-877907c944b2.jpg>

Hi Clemens,

Thanks for the detailed answer. I'm pretty sure, the touch sensor behaves
like a switch. Did you try to touch it on and then off after a second or
so? Does it stop then to increase or decrease volume?

habersatt, 2019-01-28:
Hi burberius,
that works! Thanks a lot!!!!
LG
Clemens

MiczFlor, 2019-01-28:
Hi @habersatt 
I close this thread and started a page in the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Capacitive-touch-sensors
Possibly you want to add more info to this page?
Thanks @comdoxx and @burberius for leading the way!
All the best, micz

closed 2019-01-28 by MiczFlor