MiczFlor/RPi-Jukebox-RFID #457: Maybe suggestion for double swiping - restart playlist when remember positon is set on

derOger, 2019-01-24:
Hello,
maybe a function would be good that when you swipe a card a second time the playlist restarts when the option for position rememberance (?) is on. So when the kid swipes his favourite audio book and it starts in the middle because the box remembered its position it can reswipe and the playlist starts from the beginning.
Greetings

comdoxx, 2019-01-25:
This is actually implemented.
If you open the box's website on the "Settings" page, you can pick what to do on second swipe:

- Toggle Play/Pause
- Restart Playlist from the beginning
- Skip to next track
- Ignore (does not apply to system commands such as 'volume up')

derOger, 2019-01-25:
Doesn't worked for me. He resumed the playlist - but I will try today evening again - maybe I messed up something ...

MiczFlor, 2019-01-25:
Hi @derOger @comdoxx 
Oger's suggestion actually is not implemented. At first glance this makes sense. But at second thought it might create a lot of tears - because the second swipe would would always jump back to the beginning and the stored position will be lost forever.
Have a look at the possibilities you have on the settings page for the second swipe.
All the best, micz

MiczFlor, 2019-01-26:
Another thing @derOger : combining second swipe restart and resume on folder level can create seemingly funny but „logical“ behaviour:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#settingssecond_swipe

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>