MiczFlor/RPi-Jukebox-RFID #460: Suggestions on how to solder onto my hifiBerry board

georgbachmann, 2019-01-27:
Now I've got my RC-522 as well as my hifiberry board working separately (thanks to all the super nice people here!!!)
Now to get both working at the same time, I'd have to solder the RC-522 to the hihiberry board (as that one directly covers up all the pins of the raspberry when but together)
I guess this setup shouldn't be that uncommon here? Does anybody have tips on how to solder the additional cables to the hifiberry board? Pics would be awesome. I am just not sure if I should try to get a female connector thingy (i don't know the real word for that :) ) onto it, or directly solder the cables to the RC-522?!?

romanbalsiger, 2019-01-27:
i had the same problem. not for the reader, but for some buttons and leds. i made a matching "Connector"-pHAT to fit on top of the hifiberry, as i didn't want to solder on to the hifiberry.

[image]
[image]
[image]

georgbachmann, 2019-01-27:
That looks cool!!! I see! On my hifiBerry I'd need a bit more spacing, but thats nice!
On your second image... how do you call the adapters on the left of the board? The female to male converters? I'd have to look if the exist with more spacing... then I could make a shield like the one you have! That would be ideal... cause then I can do all the wiring/soldering there and just plug everything together. I like the idea!! :) Thx!!!
Oh... and one more... what is the connection you have going directly from the raspberry to your "Connector-pHAT" (blue/white cable)?

romanbalsiger, 2019-01-27:
https://www.banggood.com/5pcs-40Pin-Single-Row-2_54mm-Round-Female-Header-Pin-p-1028334.html?rmmds=myorder&cur_warehouse=CN
https://www.banggood.com/10pcs-40Pin-2_54mm-Female-Header-Connector-Socket-For-DIY-Arduino-p-945516.html?rmmds=myorder&cur_warehouse=CN

wich hifiberry do you use? depending on the model the gpio pins on the board are much longer and you can use the longer headers.

the connection ist just usb for my reader. there was no space for proper usb connector.

[image]

georgbachmann, 2019-01-28:
I will use the Mini-Amp... I just ordered some of those pins you listed and wills stack 3 of them. That should give enough clearance :)
Thanks again for sharing your build! I think it's a cool and clean solution what will leave my hifi-berry as is :)

MiczFlor, 2019-02-14:
Hi @georgbachmann @romanbalsiger 
thanks for your contributions! I added this info as a wiki page here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/HiFiBerry-with-hardware-soldering
... and will close this thread. If you have more info, please add it to the wiki.
Thanks again, micz

closed 2019-02-14 by MiczFlor