MiczFlor/RPi-Jukebox-RFID #461: After end of playlist and option "Skip to next track" second swipe is not working

eric1905, 2019-01-27:
I have folders with multiple tracks and also folders with just one track.
Currently I have the activated the option for second swipe: Skip to next track.
If my daughter listens the card with just one track inside it, and the track ended she wants to rehear it.
Therefore she swipes the card again. The systems does not react.
I assume this is because of the second swipe option. It wants to play the next track which does not exist, so nothing will be played.
Is it possible that the system recognizes that the playlist is ended because there is no track in the playlist and then replays the whole folder?
There is a workaround to start playing another card. But this is just a workaround in my opinion. The user friendly way was to replay the playlist, even if it is not the selected option for second swipe.
Would be great if this could be implemented :)

eric1905, 2019-10-24:
Is there any possibility to implement this? would be great :)

jo-me, 2020-06-16:
The behavior above could be easily changed if desired now that https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/914 was fixed in 2.0.  

in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/rfid_trigger_play.sh
PLAYPLAYLIST=skipnext needs to be PLAYPLAYLIST=yes if the playlist is over.