MiczFlor/RPi-Jukebox-RFID #467: Texts missing in Web UI after a fresh install - how can I resolve this?

matthias2468, 2019-01-31:
Hello,
I have followed the automatic installation process for the non-Spotify version. Something must have gone (only slightly) wrong - the texts in the web interface are missing. In a lot of spots this is just ugly, but the frontend for assigning RFIDs is quite unusable.
What can I do to get this resolved?
[image]
Thank you for your advice!
Matthias

MiczFlor, 2019-02-01:
Hi @matthias2468 
could you please paste the result of the following command here?
~~~
cd
ls -l RPi-Jukebox-RFID/htdocs/lang/
cat RPi-Jukebox-RFID/htdocs/inc.langLoad.php | grep langDef
cat RPi-Jukebox-RFID/htdocs/inc.header.php| grep langLoad
cat RPi-Jukebox-RFID/settings/version
~~~

matthias2468, 2019-02-01:
Hello,
Already the first command fails. There is no folder lang under htdocs.
Best regards
Matthias

MiczFlor, 2019-02-01:
That's strange, the files are in the code base:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/htdocs/lang

Did you do the install script or the download version (which is a bit out of date).
If you haven't done anything yet but install it, I recommend you do the script again:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#one-line-install-command

I had an issue with the language versions before, but that has been fixed a week ago. Possibly your code is still from then? Please try this and paste the result:
~~~
cd
cat RPi-Jukebox-RFID/settings/version
~~~