MiczFlor/RPi-Jukebox-RFID #469: Folders with [ ] can't be expanded in Web UI

Yordan1976, 2019-02-01:
I use to name my album folders with the year of the album like "Creed - [2005] Greatest Hits".
These cannot be expanded to view the files. If I remove the brackets ("Die Toten Hosen - Laune der Natur"), everything is fine.

[image]

RogoTe, 2019-02-02:
Already reported and should be solved in next update
#33 #331 #382

MiczFlor, 2019-02-03:
Hi @Yordan1976 
please try the latest code (using `git pull`). Should work now:
#471

Yordan1976, 2019-02-03:
Sorry, now it expands the first folder if I try to expand any other.

MiczFlor, 2019-02-04:
Hi @Yordan1976 
thanks for testing. Could you try again please with `git pull`. Now it should work. I made the mistake not to test enough... :/

Yordan1976, 2019-02-04:
It works, thanks for fixing!

Yordan1976, 2019-02-11:
I'm afraid we have to reopen this issue. There are similar bugs with other special characters:
- a '+' or '.' in the folder name shows the same behaviour as the '[]' before
- a ',' in the folder name shows also the same behavior, unless there is a folder identical named before the comma. Then it expands that folder
Example:
- This is a great album
- This is a great album, Vol. 2
This Volume 2 will expand (despite the dot), but the first one.

miohna, 2020-12-23:
I came across the same issue, `()` in my case.

While looking into the details of the issue, I started wondering why we are using such a complex ID for the `collape` rather than just a number. When checking the git history, I noticed that the relevant code section was changed in Sep 2018 (1d310b56b294a75049a91b051879e3b7d3d37bde). Before, variable `$idcounter` was used. This variable still exist, but is unused now. 

@MiczFlor Is there any reason, not to use the numeric ID value?

I suggest to go back to the old numeric ID to avoid issues like the once discussed here, see #1231. 
However, I am not having full overview of the project and may miss a point/feature here.

closed 2021-02-23 by MiczFlor