MiczFlor/RPi-Jukebox-RFID #478: managing several boxes to use same rfid cards

kawuppdich, 2019-02-08:
Hi,

first of all thanks for this project. 
I have built two boxes and now  I am struggeling on what is the best and easiest way to manage the rfid cards to work on both boxes by registering them on only one box. Is there a way? What would be needed additionally, A NAS is available if needed.

Regards

Klickmichel, 2019-02-08:
Hi,
you could try to sync the 'shared' folder with rsync, triggered by a cron job.
I'm not sure, if it would be a problem to sync also the 'folder.conf' files or if they better should be excluded in rsync.

MiczFlor, 2019-02-10:
Hi @kawuppdich @Klickmichel 
there are two scripts I used - quite a while ago - to do exaclty that:
* create a "backup" of assigned RFIDs to folders
* recreate the same setting on a different Phoniebox
Have no Idea if they are still working, but please have a look:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/helperscripts/AssignIDs4Shortcuts.php
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/helperscripts/CreateCsvFromShortcuts.php

s-martin, 2020-03-07:
This is an explanation in German: http://splittscheid.de/phonieboxen-syncen/

ZyanKLee, 2020-03-07:
i currently use rsync as well - that works fine but is a manual process.
I'm thinking about switching to a more automated way by using something like syncthing or even the sync tool that comes with my NAS. The latter one would have the benefit that the sync would also work asynchronously (data will be backed up to the nas and when the other box comes online it will get its updates from the nas)

AlvinSchiller, 2023-05-04:
This feature has been added in Version 2.5.
-> See https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Sychnronising-Phonieboxes-in-a-local-network