MiczFlor/RPi-Jukebox-RFID #479: Web-UI Stop Playout Timer Value not shown in circle

SpookyFishes, 2019-02-08:
I am using phoniebo software 1.1.9-rc7 Spotify edition
 
When I select a value for the "stop playout timer" an press "Set" no value is shown in the circle.
It doesn't care if I select 2 min oder 240 min

[image]

Yordan1976, 2019-02-10:
There are more issues with the shutdown settings:
- If you set the sleep timer, both the sleep timer and the stop playout timer circles will be set.
- Both select dropdowns show "OFF" after setting the one or the other.
- If you set the idle timer and play just for a few seconds, it doesn't reset to no idle.

MiczFlor, 2019-02-14:
I committed the changes to the latest `develop` branch. ... and will close this for now.

commit f30ed80d (2019-02-14): (maint) fix #479
commit 60edab78 (2019-02-17): (maint) fix #479 and #484

First remove playerstop timer if existent, then add the new one.
commit ce1e3a7b (2019-02-17): Merge pull request #489 from Yordan1976/develop

 (maint) fix #479 and #484
closed 2019-02-14 by MiczFlor