MiczFlor/RPi-Jukebox-RFID #480: No audio - new installation

Maggal76, 2019-02-09:
Hi there,

in advance: this is my first project using a pi. I found the vids from we love tech on youtube and simply started. I followed the instruction showed in the vids. But sadly, the programm isn´t working...

I´m using Raspbian GNU/Linux 9.3 (stretch) and the 1.1.9-rc7 plus edition. No USB-soundcard. Sound is working. I can open the interface and read out RFID-Numbers. I paired those cards with local stored mp3s and a spotify playlist (the cover has been saved on the pi, but isn´t showing on the interface). But I don´t manage to run the files. By pushing the play-button in the interface: nothing happens. But putting the card over the reader: nothing happens.

Any idea, what went wrong?

Thx for your help!

Yours
Markus

Klickmichel, 2019-02-09:
Hi,
I had the same problem with new added mp3. Enter „mpc update“ in the terminal. This helped for me.

Maggal76, 2019-02-10:
Thx a lot!
That point was part of the problem! The other one was the volume ;-).
Now I just have to work out, how to connect with spotify/iris...

closed 2019-02-10 by Maggal76