MiczFlor/RPi-Jukebox-RFID #483: Feature request: Amazon music

Alwiz79, 2019-02-12:
Hi, 
is there any chance, that we will see phoniebox playing music from amazon music unlimited (instead of spotify).
I found the following on GitHub, but I guess I don't have the skill to somehow implement this to phoniebox: https://github.com/Jaffa/amazon-music

Has somebody running amazon music on a raspberry and could contribute some experience ?

makhau, 2019-02-13:
Hi,
would be interested in this too. Can't find much in the internet about Amazon prime and unlimited support for music services beneath amazon hardware or andoird phones.

MiczFlor, 2019-02-19:
Hi @Alwiz79 @makhau 
I added the `enhancement` tag to this thread. I have no amazon music account, so I don't know how to help right now :)

s-martin, 2020-04-12:
From my point of view the following steps would be necessary:

- Create a Mopidy backend (maybe based on https://github.com/Jaffa/amazon-music); guide for how to create a Mopidy extension: https://docs.mopidy.com/en/latest/extensiondev/
- Ask maintainers of Mopidy-Iris (https://github.com/jaedb/Iris) to integrate the new Amazon Music backend
- Add new dependencies to Spotify edition installer (trivial)

Overall this is a non-trivial task as Amazon Music doesn’t provide a official SDK or API and this is all reverse engineered; currently the existing amazon-music Python library seems not to work anymore (see https://github.com/Jaffa/amazon-music/issues/30) as Amazon probably made changes. In addition that means a continuous maintenance effort (to fix changes by Amazon) is necessary.

Maybe other users of Mopidy are interested, but I can’t quantify that.

s-martin, 2024-05-17:
Amazon provides the [Amazon Music Device API](https://developer.amazon.com/de/docs/music/API_browse_overview.html), but it’s already deprecated.

Not sure, if the new [Amazon Music Web API](https://developer.amazon.com/de/docs/music/API_web_overview.html) provides audio playback.