MiczFlor/RPi-Jukebox-RFID #485: PI 1 Model B - Spotify version - Slow and volume cannot be set to > 0

dvalob, 2019-02-14:
Hi!

First, thanks for your super work!!! It is exactly what I was looking for.

I used an old RPI 1 Model B that I had in a drawer. I installed yesterday the Spotify version but does not work for the moment. I put the sample audios but unable to heard something through the jack output.

The WebInterface was veeeery slow (around 30s to connect once the address typed on the browser). The volume was always 0, impossible to set a different value. I tried to change the conf data through ssh but without success.

My question is: is this model too old? I read in the README that it should work with this model 1, but did someone try? Maybe I can start by using the non-Spotify version.
My PI CPU load was continuously close to 90% :(

Thanks in beforehand!

MiczFlor, 2019-02-14:
Hi @dvalob 
I am running the "Classic" version (without Spotify) on the same machine. I also have some web UI issues - and I assume it is partly my making when I changed the icons from font awesome to material icons - which now reads 2MB of fonts when the Web UI loads. My daughter is happy, still, she uses the RFID chips and all works fine for her.
Generally speaking, there was also a drop of speed when the spotify version was introduced - and that affected the non-spotify version, too. Read here for more info: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/477

MiczFlor, 2019-02-14:
Hi @dvalob 
me again :) the volume change is strange. Did you install using the one line install script?
I suggest you do a new install with the one line install for the classic version. Once this is working, you can later try the spotify install. There are a couple of changes required for spotify that might add to the slow down. 
So I suggest to get it up and running first. Frankly, the spotify is not really needed if you do this for your kid :) there is enough on YouTube and in your local library to keep it all "Classic" and fun still...

dvalob, 2019-02-15:
Thanks for your quick reply @MiczFlor . I will try this evening again. Do you have any recommendation for a clean uninstall? I did not see it in wiki (I know, nobody wants to unistall this super project, but it could help :p ). Otherwise I will simply format the card and install again Raspbian.

Thanks again

MiczFlor, 2019-02-15:
Hi @dvalob 
actually, just run the one line command on top of the existing install. This way you can even try if the install recognizes the existing install and asks to keep configuration and audio files.

dvalob, 2019-02-15:
I will try yes, but I will overwrite everything to start from scratch.
thanks!

MiczFlor, 2019-02-19:
Hi @dvalob 
silence is golden, so I close this for now :)

closed 2019-02-19 by MiczFlor

dvalob, 2019-02-19:
HI!!

After tenths of tries I have one conf that is working. Most is working now
(i tried different raspbian versions, and for sure the one you gave in the
manual is working, but after 4 or 5 from scratch tries).

During one line install I have  alot of 404 errors and webapp was not
working, etc.. Yestarday I tried again and no more 404 errors and worked. I
used alsamixer to activate the internal pi soundcard, maybe that unblocked
the issue.

Now I am putting RIF cards and works fine. I am only having troubles to
play streams (even the sample Bayern2, even if it it is on air currently )

I will continue trying, enough for today.

Thanks!!