MiczFlor/RPi-Jukebox-RFID #486: Is it possible to have the GPIO buttons only play for 20 seconds and then automatically pause?

Shiramechanic, 2019-02-15:
I have made a Phoniebox with GPIO buttons as well as playlist specific GPIO buttons. Is there a way to put a timer on the playlist specific GPIO buttons so that pushing the button will only play a song for 20 seconds and then the song will automatically pause? The button would need to be pressed again to play each additional 20 seconds. (The purpose of this feature is to use the Phoniebox as a tool to help children with physical disabilities learn motor skills. Pressing the buttons more frequently would be necessary, as the purpose of the music is to motivate the child to press the button.)

m-rl-n, 2019-02-18:
Hi @Shiramechanic 
I have not tried, but I think this should work:

Extend `/scripts/playout_controls.sh` with a new option.
```
playerpauseafter)
        # pause player after $VALUE seconds
        echo "sleep  $VALUE ; mpc pause" | at now
        ;;
```

Then you can add `check_call("./scripts/playout_controls.sh -c=playerpauseafter -v=20", shell=True)` to the function that triggers your playlist button.

MiczFlor, 2019-02-19:
Hi @Shiramechanic @m-rl-n 
I like the thinking. And @m-rl-n is very close to a working solution. To think the use case through, I guess it would best work like this:
* swipe a card to start a playlist (a "normal" one)
* swipe the special card: "play ... then pause after 20 seconds"

This "special" card is not liked to any particular folder, it would word for anything. The matching command would look like this:
```
playerpauseafter)
        # pause player after $VALUE seconds
        echo "mpc play; sleep  $VALUE ; mpc pause" | at now
        ;;
```
I am using @m-rl-n `at now` to push the command into the background. I guess that should work, haven't tried it.
@Shiramechanic please give us feedback if and how that works.
All the best, micz

Yordan1976, 2019-02-19:
Another idea for this: A folder option to pause after one track.
In combination with shuffle and a folder full of animal sounds that would be a nice guessing game for the little ones.

Shiramechanic, 2019-02-22:
Thank you so much for your help! I added the command that you mentioned, but because I wanted it to work for all my playlist GPIO buttons I added the code into playlistaddplay in  playout_controls.sh and it now automatically pauses after 20 seconds.

 If I wanted to be able to play 40 seconds by pressing the playlist button 2 times sequentially, rather than having to wait for the timer to end before pressing the button again to reactivate the music, how do I go about that? 

Many thanks for all of your help! I am looking forward to sharing pics of my Phoniebox once it is complete!

Best, Shira

MiczFlor, 2019-03-10:
Hi @Shiramechanic 
as @Yordan1976 already pointed out: the "single track" option might also be a way to get some motion into the listening. I added this with the latest 1.1.9 release: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/507
Personally, I would be interested to hear more about the use case you are suggesting, if you feel like sharing, please mail me at micz.flor AT web.de (mail is also linked on my profile). My current day job is in the field of psychiatry and I had no idea that Phoniebox could somehow touch the issue of opening up new opportunities for people with disabilities. Thanks for opening another door to this project.
All the best, micz