MiczFlor/RPi-Jukebox-RFID #487: Problems with GPIO Buttons

georgbachmann, 2019-02-16:
On my Pi-Zero I am using a hifi berry zero as well as a RS - 522.
I how have wired GPIO Buttons to the rasbperry which should not collide with neither the hifi-berry nor the RC-522. I am using the following setup:

```
Next:        GPIO4  -> Pin 7
Prev:        GPIO17 -> Pin 11
Start/Stop:  GPIO27 -> Pin 13
Volume Up:   GPIO22 -> Pin 15
Volume Down: GPIO23 -> Pin 16
```
And I modified the `gpio-buttons-py` acordinly:

```
...
vol0 = Button(13,pull_up=True)
volU = Button(15,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(7,pull_up=True)
prev = Button(11,pull_up=True)
halt = Button(13,pull_up=True)
...
```
In the phoniebox-web-ui both RFID and GPIO-Buttons are enabled.

But for whatever reason, now the RFID isn't working any more and also the gpio buttons are not reacting at all :(

For the wiring. I just connected the one side on all buttons to the same ground on the raspberry, which should be ok, right? And the other sides like described above. No resistors need, right?!?

Any hints on how I can debug the issue would be more than welcome!!! :)

Thanks a lot!

EDIT:
so... I figured out, that when I don't use the Pin Numbers, but the GPIO Numbers in the script, I can now increase and decrease the volume. Which is a good start :)
But now the RFID-Reader is not working any more?!? I don't get it. I thought I had to put in the Pin Nr. not the GPIO Number. I am confused now :)

So with this config my RFID Reader and sound is working properly, but of corse my buttons aren't:

```
vol0 = Button(11,pull_up=True)
volU = Button(29,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(31,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(16,pull_up=True)
prev = Button(15,pull_up=True)
halt = Button(13,pull_up=True)
```
I'd be super happy if somebody could tell me what to change...
I wrote a little sample python tool to check the buttons... they are working as expected... but I am just not sure how to combine everything now :)

When running the gpio-buttons.py with those settings directly I get the following:

```
pi@phoniebox:~ $ python RPi-Jukebox-RFID/scripts/gpio-buttons.py 
/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py:245: PinNonPhysical: no physical pins exist for GPIO29
  'no physical pins exist for %s' % repr(self)))
/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py:245: PinNonPhysical: no physical pins exist for GPIO31
  'no physical pins exist for %s' % repr(self)))
```

martinclausen2, 2019-02-16:
Hi @georgbachmann yes, the GPIO numbers are the one that go into the scripts. Please adjust your script and the wiring according to this list: [Flexible Pinout](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pinout) Let us know how it goes.

georgbachmann, 2019-02-16:
Awesome... Seems I also had to comment out the unused commands. Now it works. Thanks again to all of you who helped me get the whole thing working! It's really a great tool!!!! Looking forward to my sons eyes when he's going to see it :)

closed 2019-02-16 by georgbachmann