MiczFlor/RPi-Jukebox-RFID #488: Cannot register RFID reader

Chapuisat, 2019-02-16:
Hi all,

I struggle since hours with setting it all up from scratch, especially with the RFID reader, which is the following: **Reiner SCT cyberJack RFID basis**. This device works correctly with MacOS and Windows and even under Raspbian I am able to make it work but not with the RPi-Jukebox-RFID.

When I enter **lsusb** I get the following:

`pi@raspberrypi:~ $ lsusb`
`Bus 001 Device 005: ID 0d8c:000c C-Media Electronics, Inc. Audio Adapter`
`Bus 001 Device 004: ID 0c4b:9102 Reiner SCT Kartensysteme GmbH cyberJack RFID basis contactless smartcard reader`
`Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. `
`Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub`
`Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub`
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

That means that the USB RFID device is recognized at BUS 001, Device 004 correctly.
When I swipe any RFID card over the device nothing happens.  

Only when I execute the following command I can use it. But once **pcscd** is terminated I stops working immediately.

`sudo pcscd -a -d -f`

My problem is that when I run the registration process of the RPi-Jukebox the RFID device is not listed:

`pi@raspberrypi:~ $ ls -la /dev/input/by-id/`
`total 0`
`drwxr-xr-x 2 root root  60 Feb 16 22:25 .`
`drwxr-xr-x 4 root root 120 Feb 16 22:25 ..`
`lrwxrwxrwx 1 root root   9 Feb 16 22:25 usb-0d8c_C-Media_USB_Headphone_Set-event-if03 -> ../event0`

Or 

`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python2 RegisterDevice.py`
`Choose the reader from list`
`0 C-Media USB Headphone Set`

Only my USB soundcard is listed, but not the RFID reader. Somehow the reader is not attached to an event id?  

Any ideas?

comdoxx, 2019-02-18:
I also had a Reiner SCT in stock. But that one will not work.
The USB device must be detected as a HID (i.e. Keyboard). Which is not the case with Reiner SCT readers.

You should invest in the proposed card reader and keep the Reiner one for other purposes. Unless you dig into pcsc-lite and combine these two projects...

Chapuisat, 2019-02-18:
Thanks @comdoxx, That's exactly how it is. 

I bought another RFID reader today from **Neuftech**. It's cheaper, it recognizes all types of RFID cards and it just works like a charm now! 

`pi@raspberrypi:~ $ lsusb`
`Bus 001 Device 005: ID 0d8c:000c C-Media Electronics, Inc. Audio Adapter`
`Bus 001 Device 008: ID 16c0:27db Van Ooijen Technische Informatica Keyboard`
`Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. `
`Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub`
`Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub`
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

or 

`pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ls -la /dev/input/by-id/`
`total 0`
`drwxr-xr-x 2 root root  80 Feb 18 23:10 .`
`drwxr-xr-x 4 root root 140 Feb 18 23:10 ..`
`lrwxrwxrwx 1 root root   9 Feb 16 22:25 usb-0d8c_C-Media_USB_Headphone_Set-event-if03 -> ../event0`
`lrwxrwxrwx 1 root root   9 Feb 18 23:10 usb-HXGCoLtd_27db-event-kbd -> ../event1`

Thanks anyway!

MiczFlor, 2019-02-19:
Hi @Chapuisat @comdoxx 
thanks for this research. I learned something new - and that's always a good thing. I added your findings to the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special
... and will close this thread for now. Thanks, micz

closed 2019-02-19 by MiczFlor