MiczFlor/RPi-Jukebox-RFID #490: Human readable card names

Yordan1976, 2019-02-16:
I read the manual here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#making-a-human-readable-shortcut-for-a-card
But that is not, what I understand under the headline. It is just the connected folder name.
What I would like to see is the possibility to give descriptions for the cards. For example: I put some RFID-Stickers in toys and would like to see something like "yellow car" or "blue turtle" instead or additional to the RFID id in the GUI.
The new Card_Assignments.txt from #477 could get an extra lines for the descriptions like
```
[231412138169]
cardid = 231412138169 <-- this is essentially redundant information, but in case one wants to change the "section" style...
description = Rote Plastikeule
uri = Tiergeräusche/Eule
shuffle = 0
repeat = 0
resume = 0
elapsed = 0
playstatus = stop
```
Also it would be helpful to create a page in the GUI, where all known IDs are listed for a better overview: 
ID - description - connected folder or special function

MiczFlor, 2019-02-19:
Hi @Yordan1976 
can you explain the use case where you would name a RFID with two names? Isn't the folder name human readable enough?
I could see the use case where you have a chip that is a "yellow car" - and you can just create a folder with that name. Why would that not be enough?
(not challenging your idea, just wanting to fully understand it)

Yordan1976, 2019-02-19:
The folder name is of course human readable, but not necessary a description of the card or the toy, where I've put a chip inside. Usually I have albums and therefore the folders are named after them.

closed 2020-01-18 by MiczFlor