MiczFlor/RPi-Jukebox-RFID #491: Problem to restart powerbank with OnOff SHIM

fred-h20, 2019-02-17:
I have a problem with my powerbank and the OnOff SHIM:
When I shutdown the Raspberry Pi and wait for 30 seconds, I cannot start again with the button connected to the OnOff SHIM. I have to press the button of the powerbank or replug the Raspberry Pi. After 30 seconds, the Powerbank goes to a sleepmode. The output voltage is reduced to 3 V. It detects the load with other devices and starts again. But I cannot reboot the Raspberry Pi. The button of the powerbank is not accessible from the outside of my Phoniebox so I would like to find a solution to wake up the Powerbank with the button of the OnOff SHIM.

Powerbank: EasyAcc 20000mAh ( https://www.amazon.de/dp/B01422TC14 )
OnOff SHIM ( https://shop.pimoroni.de/products/onoff-shim )

splitti, 2019-02-17:
Hi,

Try to change the USB Ports. I have also the Shim and the EasyAcc. Works finde, but i read some where, that you have to use Port One or three, the other two are somethink of sub-Ports.

Hope this helps.

Yordan1976, 2019-02-17:
I have the same stuff and use the port near to the lamp. Works fine! But I'm going to make a small hole in my box to reach the button with a paperclip. Just in case.

fred-h20, 2019-02-17:
That was the problem. It works with ports #3 from the lamp. Thank you very much.

fred-h20, 2019-02-17:
Still sometimes it doesn't work. Perhaps the micro-usb-cable is a bit loose and does not have a proper contact.

Yordan1976, 2019-02-17:
Try just to wait a minute. I had some problems to repower it, after I made some quick power one and offs for testing reasons. It recovered after a while.

fred-h20, 2019-02-18:
You were right. It works again after a few minutes. Strange. Thanks!

closed 2019-02-18 by fred-h20

Benzinbruder, 2019-12-11:
> 
> 
> Hi,
> 
> Try to change the USB Ports. I have also the Shim and the EasyAcc. Works finde, but i read some where, that you have to use Port One or three, the other two are somethink of sub-Ports.
> 
> Hope this helps.

Thanks splitti, that helped a lot!