MiczFlor/RPi-Jukebox-RFID #492: Webinterface very slow

georgbachmann, 2019-02-19:
Since today the web-ui is super slow for me :( Everything was working fine but without any (knowingly) changes that I made.
It works fine when I restart my rasberry for about a minute or so (didn't do time-tracking). After that, jumping from one page to another (f.e. file-setup to home) will load forever.

Also I just uploaded some files into a folder and it will not show in "Home". It will show that there is my folder with the subfolder. Also that there are 30 files in there. But it will not show the content when clicking on it?!?

Any idea on how to debug the problems?

MiczFlor, 2019-02-19:
Hi @georgbachmann 
that doesn't sound good. And it sounds, well, as you say: unexpected.
* are you running "classic" or "plus spotify"?
* which version are you running?
* can you ssh into the RPi and type `top` in the terminal to see what applicatoin squats the CPU power?

georgbachmann, 2019-02-19:
Hey @MiczFlor 
I am using classic and just did a `git pull` today... so latest master version
I did a `top` earlier, but didn't see anything significant. the `gpio-buttons.py` was using the most cpu with about 2-3%
Could it have something todo with the files I uploaded earlier? That it is processing something? But then the cpu would spike... hm...

MiczFlor, 2019-02-19:
Hm. Let's see if somebody else jumps into this thread with a similar issue.
Usually the RPi gets slow if 
* the power supply is not powerful enough (needs 2.5A and over)
* the RAM is full and the RPi starts swapping to the SD card
* the RPi gets too hot

georgbachmann, 2019-02-19:
Hm... mine has 2.1W (it's an iPad charger)
About the RAM... there is nothing but the phoniebox on the pi... So I guess that should not be the issue?
And at least it does not feel too hot... is there a way to check that more precisely? :)

m-rl-n, 2019-02-20:
I have similar problems sometimes. I attributed it to Wi-Fi problems because for me there are no performance issues with the ui when the pi is in a room with good Wi-Fi reception.

helloiamgary, 2019-03-04:
I'm having the same issue to the point where my phoniebox times out before it ever shows me the UI, I've put this down to having a large MPD database, the existing cards I have programmed for RFID work but I can't add any more. Connecting to MPD via another app on my laptop works fine though. 

Top comes back with the following 
16585 www-data  20   0  123812  12656   5636 D    4.6  1.3   0:40.65 php-fpm7.0 
16185 www-data  20   0  128024  16348   5828 D    3.0  1.6   0:55.99 php-fpm7.0 
466    root             20   0  227088  50324  15288 S   2.6  5.1   25:15.25 mpd        

Memory shows
KiB Mem :   994108 total,   462460 free,   172620 used,   359028 buff/cache
KiB Swap:   102396 total,   102396 free,        0 used.   729056 avail Mem 

I'm wondering if it's a PHP issue (only guessing), as I have one network share added to my Phoniebox library that has hundreds of folders in it, up until adding this the UI worked perfectly with locally created folders.

MiczFlor, 2019-03-08:
Hi @georgbachmann @helloiamgary @m-rl-n 
I had a look at this issue again. Using `top` I identified that the gpio-buttons.py (which I had switched on, but am not using - I have no buttons) took 75% CPU on my RPi 1. 
In the web UI under `settings` I switched it off and got the feeling that the webUI loads much faster. It's a feeling only :) So I wanted to share this with you and have this tested by you - and share with me *if you feel the same*

jo-me, 2019-12-30:
@MiczFlor I'm seeing the same issue with the gpio buttons daemon on my Pi1. Disabiling the service helps.
Two questions though
1) The service is automatically started on boot. how can I disable the service permanently?
2) Is there a fix for the high CPU for buttons? Are buttons not recommended for the Pi1?

Thanks!

s-martin, 2023-04-29:
Closing as this is very outdated now.

closed 2023-04-29 by s-martin