MiczFlor/RPi-Jukebox-RFID #493: How can I activate StopafterX by default?

Worren, 2019-02-19:
How can I activate StopafterX by default after startup without using a rfid card? 
I would like to automatically stop playing after X minutes.

Thanks!

m-rl-n, 2019-02-20:
Hi @Worren 
you can register a service in systemd, that runs payout_control.sh at startup.
**But** this will start the timer regardless of wether you start playing audio or not.

Here is how to do this:

- Create a new service: `sudo nano /etc/systemd/system/phoniebox-stoptimer.service`
- Paste this into nano, change the value after `-v=` to the amount of minutes the timer should run and save:
```
[Unit]
Description=Phoniebox Startup Stop Timer
After=network.target iptables.service firewalld.service phoniebox-startup-sound.service

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/usr/bin/sudo /home/pi/RPi-Jukebox-RFID/scripts//playout_controls.sh -c=playerstopafter -v=5

[Install]
WantedBy=multi-user.target
```
- Enable the service: `sudo systemctl enable phoniebox-stoptimer.service`
- If you want to disable it sometime: `sudo systemctl disable phoniebox-stoptimer.service`

Worren, 2019-02-20:
Works great. Thank you very much!

closed 2019-02-20 by Worren