MiczFlor/RPi-Jukebox-RFID #494: Raspberry PI configuration, Spotify titles and delete folders

Chapuisat, 2019-02-19:
Allrigthy, now I have the Jukebox running and it's doing great!

Nevertheless there are two minor issues I would like to address here:

1. Once RPi-Jukebox-RFOD is installed on the Raspbian OS I cannot open the "Raspberry PI configuration" any more. Is this normal behavior?

2. When I add albums and/or playlists from Spotify Premium the **title** in the web fronend and the the file title.txt in the folder is blank. Not always but in most cases. E.g. for 

album: `spotify:album:4qkVPmVQZfS3bHamToqAjM`
playlist: `spotify:playlist:3tKCZuhj0uGnYZZwCrQYoD`

here it works:
playlist: `spotify:playlist:4o6HxY0nrDjbUHFu7OnjR9`

3. How can I delete folders from the web frontend or from the smb connection? All files are read-only.

Any ideas?

Chapuisat, 2019-02-21:
1. Solved by reinstalling Raspbian from scratch.
2. When I take the Spotify URI from mopidy the user information is missing. Taking it from the native Spotify client the user is added to the URI. Not sure this causes the problem of missing titles. But now it seems to work with this approach.
3. Would be great to be able to delete audio files and folders directly from the web app

s-martin, 2020-04-12:
2. was fixed recently. Will be available in 2.0
3. deleting files is possible, deleting folders is tracked in #380 

@Chapuisat, feel free to reopen the issue, if one of the problems still persist.

closed 2020-04-12 by s-martin