MiczFlor/RPi-Jukebox-RFID #496: Feature request: Manage Phoniebox by RFID cards - button-less modus

Chapuisat, 2019-02-21:
Hi,

having the box set up since few days the follwoing idea came into my mind (and I am not sure if it was discussed before, but I was not able to find it here):

I think of a box with **no buttons**, rather than to **control the box only by RFID cards**. Which means in detail: All operations like volume up/down, next/previous track, on/off etc. is executed by a dedicated RFID card in the same way like selecting a playlist. This should be possible.

Then you have a card for each operation and the hardware effort to build in all the buttons is gone.

What do you think about this?

m-rl-n, 2019-02-21:
I think turning on the box by RFID is not possible. 
For other functions look here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#cardcontrol

Chapuisat, 2019-02-21:
thanks @m-rl-n.

Turning on the box by a RFID card might be tricky, indeed. But all the other functions described above should be possible. Using the web app is not a solution for small kids, but playing with RFID cards to control the box sounds more than reasonable for me :-)

m-rl-n, 2019-02-21:
Hi @Chapuisat 
I used the RFID card controls on my first Phoniebox prototype witch had no GPIO buttons. They worked fine. If you need any help configuring them just leave a comment

Chapuisat, 2019-02-21:
Sure, I'm interested!

MiczFlor, 2019-02-21:
The control you are looking for is such a good idea that it was built in from the start. You can even switch wifi on and off via RFID cards. please read here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#cardcontrol

Chapuisat, 2019-02-21:
Ahhh... now got it. I just did't recognize it's dealing about the same idea! Thanks, will have a try!

Chapuisat, 2019-02-21:
You can close this one!

closed 2019-02-23 by MiczFlor