MiczFlor/RPi-Jukebox-RFID #497: Installation DAC 5102 Chip

hailogugo, 2019-02-22:
Hi,
I waste a lot of time in the last days but it will not work.
The Sound is working as follow:
- By Startup
- with VLC
- If I installing Volumio with Generetic DAC or Hifiberry DAC

What also work is to do the Softwarecontrol from ALSA by the Webpage and of the Desktop.

Before I buy this DAC I read this information of the wiki:
External Non USB Audio DAC ES9023
[https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/External-Non-USB-Audio-DAC-ES9023](url)
I add the Software like the link below.

And I found this that is my wiring
[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/15#issuecomment-389517956](url)

I do already a request in the german Forum for Raspberry but I got not enough help. :)
[https://forum-raspberrypi.de/forum/thread/42168-phoniebox-dac-installieren-und-anschlie%C3%9Fen/?postID=370014#post370014](url)

Helperskript
`
pi@juke:~ $ sudo /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 1.1.9-rc7
*** edition: classic
*** Audio_iFace_Name: Master
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9,5K Feb 18 13:41 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1,1K Feb 18 13:41 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Feb 18 13:41 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Feb 18 13:41 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Feb 18 13:41 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Feb 18 13:41 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Feb 18 13:41 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Feb 18 13:41 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Feb 18 13:41 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 15K Feb 19 17:01 /etc/mpd.conf
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
#       mixer_control   "Master"                # optional
#       mixer_control   "Master"                # optional
        mixer_control   "Master"        # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1,7K Sep 10 10:31 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
#static routers=192.168.0.1
#static routers=192.168.1.1
option domain_name_servers, domain_name, domain_search, host_name
[... 179 lines cut ...]

Or other Test

`
audio_output {  
    type        "alsa"
    name        "My ALSA Device"
    mixer_control    "Master"    # optional
}  
audio_output {  
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC"
        device          "sndrpihifiberry"
        mixer_type      "software"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}  

`

Can somebody help me?
If you need something please let me know.

pseeger, 2019-02-25:
Try top disable Thema gpio buttons

hailogugo, 2019-03-01:
Hi tested it but I cannot here anything.

hailogugo, 2019-03-07:
@martinclausen2
I see you wiki data about DACs. but it do not work for me. I do not know what is wrong.

Please look up in this thread. I tested a lot of different things but I do not find what I am do wrong.

https://forum-raspberrypi.de/forum/thread/42168-phoniebox-dac-installieren-und-anschlie%C3%9Fen/?postID=371532#post371532

Please help me.

hailogugo, 2019-05-01:
Hi, 
I think I do off the gpio buttons service by using the off button on webpage. This will not stop the service. So my mistake.
If If I disable the service like it will add as a service in the installscript the dac works well.

My problem was that I also do a tested before with a LED in my button skript and this was not deactive from my side because I do not use the LED.
So now it works well and thanks for all your help.

closed 2019-05-01 by hailogugo