MiczFlor/RPi-Jukebox-RFID #498: Enhancement: Implementation of HD44780 LCD display (with and without I2C)

SimonChelkowski, 2019-02-22:
Hi all,

I use a HD44780 LCD display of type 2x16 (2 rows and 16 columns for characters). I started first not using the i2C interface and now also implemented a display in a second raspi with i2c. In the following I would like to share with you how I made it work in hardware and software.

The display shows the following information while a track is playing:
First line: date & time
Second line: track number & "title" where the title is the "mpc title" which is a combination of the artist name plus the track title. As in most cases this "title" is quite long I implemented a scrolling text feature.
If the music is paused it shows a message that the "Track number x is paused"
If no music is playing it just tells you "music is stopped"
In case the mpd (classic edition)  or mopidy (spotify edition) service is not running it tells you that the "service is not running"

### Guides I used
For the non-i2c version I used as guides the following ones:
Site in English: http://www.circuitbasics.com/raspberry-pi-lcd-set-up-and-programming-in-python/
Site in German: https://tutorials-raspberrypi.de/raspberry-pi-lcd-display-16x2-hd44780/

For the i2c-version I used as guides the following ones:
hardware part: https://www.instructables.com/id/Raspberry-Pi-Using-1-I2C-LCD-Backpacks-for-1602-Sc/
software part: http://www.circuitbasics.com/raspberry-pi-i2c-lcd-set-up-and-programming

### Note
Please note that the code for the i2c-version is much cleaner as it evolved from the non-i2c version which I wrote some time ago. Much more debugging bits and pieces are part of the non-i2c version. Furthermore the non-i2c version currently only works directly with the spotify version. I have not adapted it to the classic version, but you can see what needs to be done by looking at the i2c version where both versions are supported.
Furthermore note that the i2c bus uses GPIO3 which is mostly used to start and shutdown the raspi. I found a nice workaround [see here](https://raspberrypi.stackexchange.com/questions/72676/can-you-make-an-on-off-power-button-with-a-single-non-i2c-gpio-pin) which allows to use the i2c and still use a button to power on and power off the raspi. The trick is to implement a diode from GPIO3 to e.g. GPIO4, so that current can flow from GPIO3 to GPIO4 (anode goes to GPIO3 and cathode goes to GPIO4) This way you can connect the power button to GPIO 4 and have the usual power on/off functionality!

### Files
Files for the non-i2c version are:
- _LCD.py_
 main LCD script that makes use of RPLCD library (see also see circuitbasics.com link above) and
- _LCD.service.stretch-default.sample_
service file that runs the LCD main script at boot-up

The files are contained in the non-i2c-LCD.zip file
[non-i2c-LCD.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2896124/non-i2c-LCD.zip)

Files for the i2c version are:
- _I2C_LCD_driver.py_
 library needed to drive the LCD via i2c, originates from DenisFromHR (Denis Pleic) see circuitbasics.com link above
- _RPi_LCD_i2c.py_
 main LCD script that makes use of I2C_LCD_driver.py and
- _I2C_LCD.service.stretch-default.sample_
service file that runs the LCD main script at boot-up

The files are contained in the i2c-LCD.zip file
[i2c-LCD.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/2896142/i2c-LCD.zip)

### Installation
non-i2c Version:

1. Copy the file "LCD.py" to /home/pi/RPi-Jukebox-RFID/scripts/
2. Copy the file "LCD.service.stretch-default.sample" to /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/
3. Install and start LCD service
`cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/LCD.service.stretch-default.sample /etc/systemd/system/LCD.service`
4. register service by running, it will thereby start on the next boot-up
`sudo systemctl enable LCD `
5. Note to start the service instantly run
`sudo systemctl start LCD `
and to stop the service instantly run
`sudo systemctl stop LCD `
this is quite handy to know for debugging purposes

i2c Version:

1. First you need to know which I2C bus your Raspberry Pi has available on GPIOs:
`ls /dev/i2c-*`
It'll output "/dev/i2c-x", where x is your bus number. 
2. Now detect the adapter by using the i2cdetect command, inserting your bus number:
`sudo i2cdetect -y bus_number`
The I2C address of my LCD is 27. Take note of this number, it will be need in step 5.
3. if  i2cdetect is not found install i2c-tools
`sudo apt-get update`
`sudo apt-get install i2c-tools`
4. Next we need to install SMBUS, which gives the Python library we’re going to use access to the I2C bus on the Pi. At the command prompt, enter 
`sudo apt-get install python-smbus`
5. Modify "I2C_LCD_driver.py" line 22 which reads "ADDRESS = 0x27" and adapt it to your I2C address (see step 2.)
6. Copy the file "RPi_LCD_i2c.py" to /home/pi/RPi-Jukebox-RFID/scripts/
7. Copy the file "I2C_LCD_driver.py" to /home/pi/RPi-Jukebox-RFID/scripts/
8. next install and start "I2C_LCD.service"
`sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/I2C_LCD.service.stretch-default.sample /etc/systemd/system/I2C_LCD.service`
9. register service by running, it will thereby start on the next boot-up
`sudo systemctl enable I2C_LCD`
10. reboot and enjoy!
For test purposes you can use the following command to start and stop the service without rebooting
to start the service instantly run
`sudo systemctl start I2C_LCD`
to stop the service instantly run
`sudo systemctl stop I2C_LCD`

I hope you like the short guide.

Cheers,
Simon

domu83, 2019-02-24:
@SimonChelkowski that's awesome. Many thanks for posting all the details.
Maybe some comments for following users...
for the i2c version you missed to mention this step:

> Copy the file "I2C_LCD.service.stretch-default.sample" to /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/

In addition I think it will be helpful, to mention, that your files are prepared for the classic version and if the Spotify version is used there is a need to uncomment some lines in the files.

As you mentioned for the i2c version, GPIO3 is used in Phoniebox code for shutdown. If shutdown button is not used, Phoniebox code can be adjusted accordingly.

What I found is, that it seems that if I switch from a playing part to another one, it switches not instantly to the new playing.

again... MANY THANKS for providing this. This is massivly helpful

mistersunshine2020, 2020-01-16:
Hi there, thanks for your awesome work. I've managed to get this working, but for some reason my LCD is intermittently flickering when scrolling - I'm not sure why but it seems to be alternative with each character scroll with the track name, which is making it very tricky to look at. 

Is this a problem with the LCD or something I've done wrong with the code (which I copied verbatim from the zipped file)? 

Also, is it possible to have 

LINE 1 - ARTIST NAME
LINE 2 - TRACK NUMBER: TRACK NAME

I'm very new to all of this, so all help much appreciated. I've managed to get both lines scrolling by repeating the code and changing various instances to 1 instead of 2, but the flickering remains, and the output isn't running simultaneously, so both lines are alternating the scrolling. I'd be very happy with:

LINE 1 - ARTIST NAME (STATIC)
LINE 2 - TRACK NUMBER: TRACK NAME (SCROLLING)

I've connected via IC2 using 
Pin 2: VCC
Pin 3: SDA
Pin 5: SCL
Pin 6: Ground

mistersunshine2020, 2020-01-16:
> Hi there, thanks for your awesome work. I've managed to get this working, but for some reason my LCD is intermittently flickering when scrolling - I'm not sure why but it seems to be alternative [...]

I've worked out that the flickering was due to the contrast of the LCD - I've amended it via the screw on the back and it's working much better now.

I'd still love some help with the changing of information displayed on the screen though, if anybody knows?

SimonChelkowski, 2020-01-17:
Hi, 

nice to see that somebody is using my stuff :-)

The flickering is due to the fact that the display does not refresh instantly but has some after glow (is there such a word). Meaning the letters fade in and out on the old and the new position. As the scrolling is done each 0.5 seconds the letters do not stay very long on full brightness/contrast. 
You can play around with the line
`val_delay = 0.5`
in the constants section at the top of RPi_LCD_i2c.py. The number sets the time delay for the text scrolling to the next position. If you lower the number, the scroll speed increases while increasing the number lowers the scroll speed. If you set 
`val_delay = 1` you get much less "flickering" but also slower scroll speed.

Concerning your request

> ... is it possible to have
> LINE 1 - ARTIST NAME
> LINE 2 - TRACK NUMBER: TRACK NAME

Yes, of course. 

I updated the script _RPi_LCD_i2c.py_
[RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4078704/RPi_LCD_i2c.zip)

In the top section you will now find the following section, where you can set the "switches" to change the display text in the two displays rows
```
#### change the following "switches" to select what is displayed in the two display rows. 
selection_switch_first_row = 'artist' # choose from 'date_and_time' and 'artist'
selection_switch_second_row = 'track_title' # choose from 'track_title' and 'track_artist_title'
```
I hope it is easy to use enough. The file is configured currently to display 
> LINE 1 - ARTIST NAME
> LINE 2 - TRACK NUMBER: TRACK NAME
and if any of them is too long, it will scroll the text automatically.

The scrolling is done one after each other. I do not know an easy to implement solution to do it simultaneously.

Have fun with it.

Cheers,
Simon

SimonChelkowski, 2020-01-17:
Please change the value in the file _RPi_LCD_i2c.py_ (see previous post)
`val_delay = 0.1`
in line 10 to
`val_delay = 0.5`

Otherwise the scrolling is to fast. I made it that fast for debugging purposes. Sorry to have uploaded the file with the wrong value.

mistersunshine2020, 2020-02-05:
> Hi,
>
> nice to see that somebody is using my stuff :-)
>
> The flickering is due to the fact that the display does not refresh instantly but has some after glow (is there such a word). Meaning the [...]

Hi Simon, 

Thanks so much for this, and apologies it's taken so long to get back to you to offer my gratitude. I've finally got it all working and it's great - the enhancements you've provided are wonderful.

However, I've noticed a couple of bugs in the way information is displayed. 

The first is that when a song changes, it isn't clearing the screen entirely, and often characters from the previous track title are appearing on the screen. 

EXAMPLE: 

1: awesome song name
2: shorter song nameme

I think it's happening when a shorter song title isn't completely scrolling over the previous longer one. 

I've also noticed an inconsistency which may account for this - if track number is specified, I get:

1: Scrolling song title

But if the metadata contains the number of tracks on the album I get this:

1/12: Scrolling song title

I wonder if this could be messing things up? I often get track number and double colons too. 

I think the 13 character title limit is designed to accommodate 

12:thirteen char

So two digits for track number, then colon, then 13 remaining for the song. 

Finally, the length of some long songs isn't completing - I guess there's a time limit or a character limit? 
Example:
13:The Most Beautiful Girl in the Worl

It's beyond my expertise to fix this, but I wonder if a solution would be to: 

1) Use track number only, disregarding metadata about total tracks on album
2) Add a leading 0 if the track number is below 10 to accommodate the three character allocation for track number and dividing colon, leaving 13 only for the song title

Is this possible and is it a relatively easy tweak? 

Also, I think the two scrolling rows is having a negative impact on the metadata refresh for song title - it's waiting to complete the top row scroll before updating the bottom. Is there any way to keep the top row static? 

Thanks for this in advance, and apologies that this is a list of requests -  I hope that they're helpful to others (working on the assumption that this isn't just affecting me!)

Best,

Tom

SimonChelkowski, 2020-02-07:
Hi Tom, 

thanks for the feedback. I guess I corrected all the bugs you report by changing the script _I2C_LCD_and_GPIO_buttons.py_.

Please use this updated script [RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4173146/RPi_LCD_i2c.zip)

This should solve all the issues as I do not get the errors you describe anymore when using the updated file. Please provide feedback, thanks.

Please note that the track number is no longer displayed e.g. as "1/12:" but "01:" is displayed now.

Best regards,
Simon

PS: I have a further update I am currently working on. The future updated version will change the display when a button is pressed instantly. I will post this update here when I am done.

s-martin, 2020-02-08:
Would make sense to add this to the ˋcomponentsˋ directory. What do you think @MiczFlor ?

davidfri, 2020-02-09:
Hi Simon,
first of all: Thanks for your work!
I have add some changes on your code. 
[RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4176730/RPi_LCD_i2c.zip)
   - "length_available_for_title" is being calculated (one position is missing, cause i don't understand the code).
   - Display only refresh on changes, so the display doesn't flicker anymore.
   - n_cols is used everywhere, where it is posible. So it runs also on an 4x20 display.
   - scrolling text looks nicer ;)
   - bugfixes
next steps:
   - optimize code for individual sequence
   - optimize "time per circle", so all lines can scroll at the same time and update the current track don't wait until scroll is complete.

I hope you can accept the changes, so we have the same master.

Best regards

David

davidfri, 2020-02-09:
I added all features I need. Here you can see a 4x20 Display, but you can also configure 2x16 or something else:
[image]
I have to change a lot, so I think the gif shows more than thousand words. 

You can config your matrix dimensions at the following lines:
```
  n_cols = 20
  n_rows = 4
```
and the Infos that you want to show on the display could be configured at these lines:
```
#### change the following "switches" to select what is displayed in the display rows. 
info_at_line_1_play = 'artist' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
info_at_line_2_play = 'track_title' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
info_at_line_3_play = 'track_time' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
info_at_line_4_play = 'date_and_time' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 

# first line showes the state
info_at_line_2_pause = 'track_artist_title' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
info_at_line_3_pause = 'track_time' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
info_at_line_4_pause = 'date_and_time' #choose from 'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title' 'track_time' 
```
Here is the file:
[RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4178003/RPi_LCD_i2c.zip)

Best regards
David

SimonChelkowski, 2020-02-09:
Hi David,

thanks for the additional improvements. I see that you also changed the scrolling, such that it starts on the most left hand postion and scrolls from there to the left. I also changed this in my update I am working on.

I corrected the column position for the title in the scrllong text case for `selection_switch_second_row == 'track_artist_title'`.

Furthermore I added my code back to set the track number. Maybe you have not have the case yet, but if the music file played has set the _track number_ (e.g. _3_) and the _total tracks_ (e.g. _22_) then the the _track number_ read by the mpc command is _3/22_ (My code will set track number to _3_) . In the case of audiobooks this can easily become _121/231_ (My code will set track number to _121_) using a lot of the actual display space. Since I swapped from EAC to fre:ac (free audio converter) I am confronted with the fact that the _total tracks_ (or whatever it is called) is set and the display acted in the way described above. 

Hence, I recommend to keep the code between line 83 and line 100.

Best regards,
Simon
[RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4178309/RPi_LCD_i2c.zip)

SimonChelkowski, 2020-02-10:
Dear all,

as I already indicated some days ago, I hereby present to you the updated code for the HD44780 display which uses interrupts when a button is pressed to "instantaneously" update the LCD display.

**Features:**
- The buttons _prev_, _next_, _play/pause_ and _mute_ will lead to an "instant" change of the display 
- pressing the button _play/pause_ shows the track number and `music_paused_string` (default value: _paused!_) in the second display row
- pressing the button _prev_ or next shows the track number and title of the previous or next song in the cue. Simultaneously the artist is displayed in the first row for about three seconds before row one changes by th date&time.
- pressing the button _mute_ shows the `music_muted_string` (default value: _- Music muted! -_) in the first row 

**File list: (zip-file contents)** [I2C_LCD_and_GPIO_buttons.zip.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4178475/I2C_LCD_and_GPIO_buttons.zip.zip)
- gpio-buttons.py
- I2C_LCD_and_GPIO_buttons.py
- phoniebox-I2C_LCD_and_GPIO_buttons.service.sample

**Description:**
To make this work, the script needs to be executed using python 3. Hence, I ask some of you guys to test the code before installing the service (description is given below). The reason I recommend this is that I fear that a package needs to be added to you python3 version installed to your system.

Furthermore, I had to move the "code" for some of the _gpio-buttons_ to the _LCD-display_ script. Hence, I provide a modified _gpio-buttons.py_ as well. The _gpio-buttons.py_ now only contains the power button. All other buttons are part of the display script _I2C_LCD_and_GPIO_buttons.py_. 

Note: That you have to modify the files to set your specific gpio-buttons pin numbers to make the code work at all. 

When you look into the file _gpio-buttons.py_ you fill find the following comment:

```
### Initialization for Power Button on GPIO#4
### Note that I use my power button on GPIO#4 and not GPIO#3, as I use an HD44780 LCD Display connected via I2C Bus on GPIO#2 & GPIO#3. A Diode is connecting GPIO#3 and GPIO#4 allowing me to power on/off the Raspberry Pi via GPIO#4. 
#
# The following sketch uses GPIO labelling according to "GPIO.setmode(GPIO.BCM)"
#
#                 Diode       GPIO#4
#     •------------|>|----------•
#   GPIO#3    |                 |
#             |                 |
#             |                 •------•   •-----• GND
#             |                 |      •---•
#             •                 |        |
#        LCD pin: SCL           |    push button 
#                               |
#                              | |
#                              | | 10k Ohm resistor
#                              | |  
#                               |
#                               |
#                               •
#                             +3.3V
```
This shows you how to connect the HD44780 display via I2C bus while maintaining the power on and "soft"-power-off gpio-button feature!

Installation instructions for LCD & GPIO-button script:
- backup your version of _gpio-buttons.py_ before continuing, you find it in _/home/pi/RPi-Jukebox-RFID/scripts_
- copy the two python files into the folder _/home/pi/RPi-Jukebox-RFID/scripts_
- copy the file _phoniebox-I2C_LCD_and_GPIO_buttons.service.sample_ into the folder _/home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/_

- if applicable: disable your current LCD service (example name: _LCD_display.service_) by running 
```
sudo systemctl disable LCD_display.service
```
- To install the new LCD display & GPIO-button script run the following commands as also given in the file _I2C_LCD_and_GPIO_buttons_
```
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/phoniebox-I2C_LCD_and_GPIO_buttons.service.sample /etc/systemd/system/phoniebox-I2C_LCD_and_GPIO_buttons.service
sudo chown root:root /etc/systemd/system/phoniebox-I2C_LCD_and_GPIO_buttons.service
sudo chmod 644 /etc/systemd/system/phoniebox-I2C_LCD_and_GPIO_buttons.service
sudo systemctl enable phoniebox-I2C_LCD_and_GPIO_buttons
```

**Todo:**
- finish translating code to 4x20 LCD display
- review Davids code from his last post

davidfri, 2020-02-11:
Hi Simon,
I didn't looked at your code up to now, but I have a question:
Why don't you use a Software-Bus?
I added the following line at /boot/config.txt
`dtoverlay=i2c-gpio,bus=8,i2c_gpio_sda=22,i2c_gpio_scl=23`
So I can use the GPIOs 22 and 23 for the display and GPIO3 for the power-switch.
On my setup it works fine.
Best regards
David

mistersunshine2020, 2020-02-11:
> Hi Tom,
>
> thanks for the feedback. I guess I corrected all the bugs you report by changing the script _I2C_LCD_and_GPIO_buttons.py_.
>
> Please use this updated script [...]

Hi Simon, 

Thanks for the updates! It seems to work absolutely fine! I'm very intrigued by the instant change when a button is pushed, but will respond to that post separately. Amazing stuff as ever. 

Best,

Tom

mistersunshine2020, 2020-02-11:
> Dear all,
>
> as I already indicated some days ago, I hereby present to you the updated code for the HD44780 display which uses interrupts when a button is pressed to "instantaneously" update the LCD [...]

Hi Simon, 

So I'm intrigued by this, but my set-up is a little different, and I'm not sure how to accommodate your script. I've assigned 7 GPIO buttons, with four GPIO pins dedicated to the LCD display. I've only connected the buttons and nothing else - so no resistors or anything like that. My shutdown button sends a software command, and I've got a toggle switch extension on the power USB cable which seems to be sufficient for me. 

I've also used the buttons to call to different commands in the playout_controls.sh file, so I'm using the following: 

> 

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_rbt():
    check_call("./scripts/playout_controls.sh -c=reboot", shell=True)

def def_shfl():
    check_call("./scripts/playout_controls.sh -c=playershuffle", shell=True)

def def_play():
    check_call("./scripts/playout_controls.sh -c=playlistaddplay", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

shut = Button(17, hold_time=2)
rbt = Button(12,pull_up=True)
shfl = Button(26,pull_up=True)
play = Button(24,pull_up=True)
next = Button(23,pull_up=True)
prev = Button(22,pull_up=True)
halt = Button(27,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

shut.when_held = def_shutdown
rbt.when_pressed = def_rbt
shfl.when_pressed = def_shfl
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt

pause()

This means I can shuffle a folder, or restart the folder from scratch, reboot, shutdown, next, previous, play/pause. 

So given that setup, is configuring the instant display change possible? 

Best, 

Tom

SimonChelkowski, 2020-02-14:
Dear @mistersunshine2020 (Tom),

the change you ask for is possible. You can do the change on your own if you like. All you have to do is the following:

Note that in your setup, which is the classic (non interrupt) usage of the buttons, there are two individual lines and a function that belong together for each button. 
For example for the "next" button there are in your setup:
The function
```
def def_next():
	check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
```
and the code lines
```
next = Button(23,pull_up=True)
next.when_pressed = def_next
```
All you need to do to adapt to the interrupt usage version is the following:
Modify the function to be:
```
def def_next():
	check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
	global input_state
	input_state = False
	global counter
	counter = 0 # only needed in the functions for "next" and "prev" buttons, as it is needed to replace for about three seconds the date&time display in row 1 with the artist name
```
and the code lines to be:
```
GPIO.setup(23, GPIO.IN, pull_up_down=GPIO.PUD_UP)  
GPIO.add_event_detect(23, GPIO.FALLING, callback=def_next, bouncetime=400) # Note: that the bounce time depends on the physical buttons used. The bouncetime should be minimized. But if you get two "events" with one button pressing procedure, you need to increase the bounce time again.
```

Let me know if the description is usable and if you get it running for your specific setup.

I currently do not recall, why I left the shutdown button in the _gpio-buttons.py_. Just leave it there and do not change it, as I do not use this button to produce an instant update of the display as the box will turn off soon after I hold the shutdown button for two seconds

Best regards,
Simon

davidfri, 2020-02-15:
Hi Simon,
I tested the mpd-client. It is 5 to 20 times faster than the mpc/cutting option. My cyrcle-time is now 0.2 seconds (the most of the time is needed for writing on the display), before it was 0.4 seconds. I think, that I don't have many time in the next days or weeks, so I share my current state. 
Maybe you can use some of the code for your code.
Best regards
David
[RPi_LCD_i2c (2).zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4208853/RPi_LCD_i2c.2.zip)

mistersunshine2020, 2020-02-17:
> Dear @mistersunshine2020 (Tom),
>
> the change you ask for is possible. You can do the change on your own if you like. All you have to do is the following:
>
> Note that in your setup, which is the [...]

Thanks Simon, this looks fantastic. So just to clarify - I'd only need to edit the next and previous functions in the existing gpio-buttons.py script using the above, and everything else should work? 

You're a star for all your work on this. I hope others are benefitting as much as I am from your efforts. 

Best,

Tom

davidfri, 2020-02-22:
Hi @ all,
I do more changes. I think it's now a good code-base. The code has a fix cyrcle time, which could be configure. So the Display-Update (and scrolling text) has an constant interval. Also you can edit a start_stop delay at scolling text. fpr example you get these positions:
0,0,0,0,0,1,2,3,3,3,3,3,3,0,0,0,0,0,1,2,3,3,3,3,3,0,...
``` 
################# CHANGE YOUR SETTINGS HERE!!! ###########################################
## Display settings                                                                     ##
n_cols = 20                 # EDIT!!!  <-- number of cols your display has              ##
n_rows = 4                  # EDIT!!!  <-- number of rows your display has              ##
val_delay = 0.4             # EDIT!!!  <-- speed of the scolling text                   ##
start_stop_sc_delay=4                                                                   ##
##                                                                                      ##
``` 
And you can edit your individual informations 
```
################## CHANGE YOUR INFOS, WHICH WILL BE SHOWN ON THE DISPLAY #################
##                                                                                      ##
##  You can choose between the following infos:                                         ##
##  'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title'           ##
##  'track_time', 'pause_string', 'stop_string', 'mpd_not_running_string'               ##
##                                                                                      ##
##  if you have less than 4 lines, you can leave the unused lines by default            ##
##                                                                                      ##
##  Choose infos while state is "play"                                                  ##
##                                                                                      ##
info_at_lines_play [0] = 'artist' #  <-- Choose your favorite                           ##
info_at_lines_play [1] = 'track_title' #  <-- Choose your favorite                      ##
info_at_lines_play [2] = 'track_time' #  <-- Choose your favorite                       ##
info_at_lines_play [3] = 'date_and_time' #  <-- Choose your favorite                    ##
##                                                                                      ##
##  Choose infos while state is "play"                                                  ##
##                                                                                      ##
info_at_lines_pause [0] = 'pause_string' #  <-- Choose your favorite                    ##
info_at_lines_pause [1] = 'track_artist_title' #  <-- Choose your favorite              ##
info_at_lines_pause [2] = 'track_time' #  <-- Choose your favorite                      ##
info_at_lines_pause [3] = 'date_and_time' #  <-- Choose your favorite                   ##
##                                                                                      ##
##  Choose infos while state is "play"                                                  ##
##                                                                                      ##
info_at_lines_stop [0] = 'stop_string' #  <-- Choose your favorite                      ##
info_at_lines_stop [1] = 'date_and_time' #  <-- Choose your favorite                    ##
info_at_lines_stop [2] = 'nothing' #  <-- Choose your favorite                          ##
info_at_lines_stop [3] = 'nothing' #  <-- Choose your favorite                          ##
##                                                                                      ##
##  Choose infos while state is "play"                                                  ##
##                                                                                      ##
info_at_lines_mpd_not_running [0] = 'mpd_not_running_string' #  <-- Choose your favorite##
info_at_lines_mpd_not_running [1] = 'date_and_time' #  <-- Choose your favorite         ##
info_at_lines_mpd_not_running [2] = 'nothing' #  <-- Choose your favorite               ##
info_at_lines_mpd_not_running [3] = 'nothing' #  <-- Choose your favorite               ##
##                                                                                      ##
##########################################################################################
```
In "MAIN" all song-infos are getting by mpd.client, which is much faster than mpc. 
Getting all infos is now a small loop:
```
			########### GET INFOS, IF STATE IS "PLAY" #######################################
			if state == "play":                                                             #
				for row in range(n_rows):                                               #
					lines[row] = choose_line(info_at_lines_play [row])              #
			#################################################################################
```
```
def choose_line(info_text):
	switcher = {
		'pause_string': print_pause_string(),
		'stop_string': print_stop_string(),
		'mpd_not_running_string': print_mpd_not_running_string(),
		'track_title': print_track_title(),
		'track_artist_title': print_track_artist_title(),
		'artist': print_artist(),
[... 15 lines cut ...]
The delay for the constant cyrcle time is getting calculated while runtime
```
		################## EXACT CYRCLE TIME #################################################
[... cut ...]
		last_time=current_time # save time for next cyrcle                                   #
                ######################################################################################
```

I think, that your problem with the wrong tack_number is also solved by using mpd.client.

For using that code, you nedd to install a bit:
```
sudo apt-get install i2c-tools python-smbus python3-numpy python-mpdclient python-mpd 
pip install smbus numpy python-mpd2
```

Best regards

David
[RPi_LCD_i2c.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4239137/RPi_LCD_i2c.zip)

s-martin, 2020-02-22:
I think this would be helpful for many people, if you add your code to the repository.

Here is some info, how you could add your code: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/CONTRIBUTING.md

I think your code should get into the ´components\displays´ directory. If you have questions, just ask here.

SimonChelkowski, 2020-02-22:
Dear @mistersunshine2020 

> Thanks Simon, this looks fantastic. So just to clarify - I'd only need to edit the next and previous functions in the existing gpio-buttons.py script using the above, and everything else should work?

Yes, this is correct.

In case you experience any issues, log into the phoniebox using ssh and stop (temporarily until next boot) the gpio-button service by typing
`sudo systemctl stop phoniebox-gpio-buttons.service`

If you then run from within the _~/RPi-Jukebox-RFID_ directory the gpio-buttons script by typing 
`python3 ./scripts/gpio-buttons.py`
then you can test the code. In case you experience an error this is written into the console. In this case please post the output when the error occurs into this issue thread.

Best regards,
Simon

SimonChelkowski, 2020-02-22:
Dear @davidfri 

I checked your latest code and I like it very much. Although it uses polling an (infinite loop) and not interrupts (like my improved version) it works very smoothly.

I found a minor bug in the _print_artist_ function which called _loop_string_ with three instead of the allowed two arguments.

Furthermore I made some corrections in the _CHANGE YOUR SETTINGS HERE!!!_ section and corrected the spelling of _cycle_.

In my view your code is simpler to implement into the existing project than mine, as it leaves the _gpio-buttons.py_ script untouched. Nevertheless I like the flexibility of my interrupt based script more. I guess it is a personal taste. I will show my Kids your version and let them decide what will run onto their phonieboxes :-). 

[RPi_LCD_i2c_modified.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4240208/RPi_LCD_i2c_modified.zip)

Best regards,
Simon

davidfri, 2020-02-23:
Hi @SimonChelkowski 
thank you for review the code. I hope you can change to it ;)
In you code, I think the design isn't optimal.
You only respond on changes at the buttons, so if someone use the web-interface or a RFID-chip as input for "next_title", your code doesn't realize that.
While scrolling text, only the scrolling line is updating, if you also show the elapsed time of the title or something like that, it wouldn't update on the display. (With 4 lines, you get this case ;) )
And at least I think our code (RPi_LCD_i2c_modified) can also be used in all MPD-based projects, so it is a benefit for more people.
If the infos which are shown not your favorites, feel free to add a new definition ;)
I really hope, that you can switch to the "RPi_LCD_i2c_modified"-code without limitations, so we can add it to the phoniebox-main-code.
Best regards
David

SimonChelkowski, 2020-02-24:
Hi @davidfri 
I agree with you. The version in the _RPi_LCD_i2c_modified.zip_ (or a version based on it) should be the one that is added to the project. It has better, let's say, implementation compatibility and fulfils the current needs and is easy to use.

So now that this is said, I would like to inform you that the current version found in the _RPi_LCD_i2c_modified.zip_ is not fully compatible with Raspbian Strech but requires Raspbian Buster as it seems . I learned this yesterday when I implemented the code on a Raspi with a 4x20 LCD Display. The feature that is not working is the _track_time_ feature, which should display the elapsed time and the duration of the current track. For me the feature did not work at all. I wanted to have the elapsed time and duration visible in row four of the display, but row four was empty. I analysed the problem and here are my results.

- the _track_time_ feature makes use of _MPDClient.status()_ command and asks for the _duration_ (see line 238 where the command `duration=status['duration'].split(".")[0]` is used). 
- according to the [MPD documentation](https://python-mpd2.readthedocs.io/en/latest/topics/commands.html#MPDClient.status) the _duration_ feature has been introduced with MPD 0.20
- Raspbian Stretch, which I run on the Raspi with the 4x20 LCD display, reports that the most up-to-date version of MPD is 0.19.21, which is already installed. An upgrade via `sudo apt-get install --only-upgrade mpd` is not working as it reports that the MPD version installed is the latest version. I used `cat /etc/os-release` to determine the Raspbian version and `mpd -V` to identified the installed version of MPD
- Due to the MPD version below 0.20 the _duration_ parameter is unknown and results in an error which is nicely handled by the scripts exception `except KeyError:`
- The problem of not being able to update to the latest MPD on Raspbian strech is reported [here](https://www.raspberrypi.org/forums/viewtopic.php?t=223776)

In conclusion, I plan to upgrade the corresponding Rapsi with the 4x20 LCD display to Raspbian Buster.

Furthermore, I recommend that the described issue should be mentioned in the _RPi_LCD_i2c.py_ file to inform potential users, which also still run their Raspi on Raspbian Strech, about the potential incompatibility  (non working _track_time_ feature).

Best regards,
Simon

SimonChelkowski, 2020-02-25:
Hi @davidfri ,
thanks for implementing the fallback solution based on mpc. It works great. I really like the line-option _track_time_and_number_. It works nicely even for audiobooks with more than 100 tracks. 

I did one or two pull requests in the past for this project. I can try to do one on the weekend. But as far as I remember I need to get a developer version running again. I will see and let you know.

Best regards,
Simon

davidfri, 2020-02-26:
Hi @SimonChelkowski ,
I have added a few features. Now all functions are implemented, which I imagined.
There is now also a line option "title", because by using "track_time_and_number" the track number no longer has to be shown in the second line.
I also introduced state icons, which can be seen in the upper right corner:
```
use_state_icons = "yes" #choose "yes" if you want to use this ##
blinking_icons = "no" #its an add-on for "use_state_icons", so "use_state_icons" must be "yes" 
```

At first I thought that blinking icons looked nice, but I think I would like it better without blinking;)
And I add a switch off of the backlight, so that the display is switched off after x seconds when state is "stop" or "not_running".
```
backlight_off_while_waiting = "yes" #not activ in state "play" and "pause" ##
backlight_off_delay = 10 # Delay in seconds ##

```
Best regards

David
[RPi_LCD_i2c_state_icons.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4255600/RPi_LCD_i2c_state_icons.zip)

mistersunshine2020, 2020-03-09:
> Hi @SimonChelkowski ,
> I have added a few features. Now all functions are implemented, which I imagined.
> There is now also a line option "title", because by using "track_time_and_number" the track [...]

Hi David, 

I'm overwriting the RPi_LCD-i2c.py buttons with the contents of your script, changing the rows to 2 and the columns to 16. But doing so disables the display from working entirely.

Am I missing a step? 

Best,

Tom

davidfri, 2020-03-10:
Hi Tom,
did you install the new dependencies?
```
sudo apt-get install i2c-tools python-smbus python3-numpy python-mpdclient python-mpd 
pip install smbus numpy python-mpd2
```
If you already did it, run the script from the command line.
First stop the service:
``` sudo systemctl stop I2C_LCD```

Then run the script:
```
python /home/pi/RPi-Jukebox-RFID/scripts/RPi_LCD_i2c.py
```
Now you will get output, if the script run in an error. Please test ist.
I didn't test the non-I2C-Variant, so maybe, there are some changes to do. But you overwrite the RPi_LCD-i2c.py, so I think you would use the I2C-Variant ;)

You have also edit the section "INFO_AT_LINE_XXX", The default is optimized for a 4x20 display.

Best regards

David

mistersunshine2020, 2020-03-10:
> Hi Tom,
> did you install the new dependencies?
>
> ```
> sudo apt-get install i2c-tools python-smbus python3-numpy python-mpdclient python-mpd
> pip install smbus numpy python-mpd2
> ```
>
> If you already did [...]

Hi David, 

Ah, I have not yet! I will try that installation. Yes, I'm using i2c so hopefully this will make it all work. Quick additional question - I'm using Stretch rather than Buster - this has been addressed in your latest version? I know that you and Simon discussed it earlier. 

Best,

Tom

davidfri, 2020-03-10:
Hi Tom!
My last test was also on stretch, so it should have to work. Before I used buster. The latest changes were developed on stretch, but should also works on buster.
@SimonChelkowski : 
Maybe it is a good idea, if you add the current code (and addition installation instruction) in your first post. This is what people read ;)
I also think, that a pull request isn't usefull, because the label was downgraded to "hack". As I understand it, this means that this code is not welcome in the main code, so you can save yourself time.
Best regards
David

SimonChelkowski, 2020-03-10:
Hi Tom,

I use Stretch myself, it works fine. I therefore confirm David's statement. 

@davidfri good idea to edit the initial post. I will do so. As I am currently travelling I can do this earliest next week. Furthermore, I got a development system running (without display yet, need to warm up the soldering iron again :-) ). I plan to place the pull request as there is already a "display" code in the code, see code base directory _/components/displays/dot-matrix-module-MAX7219/_
We can then discuss if it wil be added to the code base in the corresponding pull request thread.

Best regards,
Simon

mistersunshine2020, 2020-03-11:
Thanks Simon and David, 

I just wanted to say how good it's been to see you both collaborating on this. It's a great enhancement, and thanks again for your efforts. 

I've got it all working - the only thing is I'm not getting track numbers at the beginning of line 2 - it looks like that information isn't available in the info choices? 

Best,

Tom

davidfri, 2020-03-11:
Hi Tom,
have you changed that line:
```
info_at_lines_play [1] = 'title' #  <-- Choose your favorite                            ##
```
to
```
info_at_lines_play [1] = 'track_title' #  <-- Choose your favorite                            ##
```
?
You can configure all infos in all states like you want!
```
################## CHANGE YOUR INFOS, WHICH WILL BE SHOWN ON THE DISPLAY #################
##                                                                                      ##
##  You can choose between the following infos:                                         ##
##  'date_and_time', 'artist', 'nothing', 'track_title', 'track_artist_title'           ##
##  'track_time', 'pause_string', 'stop_string', 'mpd_not_running_string'               ##
##  'track_time_and_number'                                                             ##
##              
```
If you are missing an option, please let me know.

In the future we need default configs for 4x20 and 2x16, but I'm currently not shure, what is the best way to realize that.
I didn't have a 2x16 display, but I thing the most useful division could looks like that
````
------------------
|Artist-SongnameX|  <- X stands for "state-icon"
|0:12/3:25  12/24|
------------------
````
Currently I'm designing the 3D-chasing and after that I have to implement a ads1115 for volume-control (or find a code ;))

Best regards
David

mistersunshine2020, 2020-03-11:
Hi David, 

Thanks for the info - so 'track_title' shows track number and track title? I didn't realise that (and not at home to test) - 'title' isn't listed in the available options so I wasn't clear what the difference would be. Perhaps a brief description for each of the options may help for people selecting in future? 

'date_and_time' - 01/01/2020 15:44 - 16 characters
'artist' - artist name from metadata (scrolling)
'nothing' - nothing
'title' - track title from metadata (scrolling)
'track_title' - track number and title from metadata - title scrolling, with leading 0 for track number if below 0

etc..

I haven't tested the various options, but I thought Simon's 2x16 solution was pretty elegant in providing the most useful info within a short amount of space: 
1: Artist (scrolling)
2: Track Number with leading 0 if below 10: Track title (scrolling)

Potential other variants could include: 

Paused state: 

Track NNN paused (which is within the 16 character limit

That would work nicely with 'track_artist_title' for line 2

I'd also like to request that the backlight off be active during pause - I don't have a stop - only play/pause, and that would help with the light pollution! 

Best,

Tom

davidfri, 2020-03-11:
Hi Tom,
you are right, "title" missed in the comment. If you have better names for the settings, let us know. My english is very bad, but I do my best that people can understand what I mean ;)
The description is a very good idea! Maybe with examples. What is the best place for it, comments in the code or a readme?

>  Paused state:
> 
> Track NNN paused (which is within the 16 character limit
> 
> That would work nicely with 'track_artist_title' for line 2

That's no problem to implementate, I will do it in the future. But I think it will be hardcoded, so translations are only possible in the code. 

> I'd also like to request that the backlight off be active during pause - I don't have a stop - only play/pause, and that would help with the light pollution!
In the future this will be implemented with a switch. 

I think the changes will be implemented in about 2 weeks (if no one else will do it in the meantime)
Best regards
David

mistersunshine2020, 2020-03-11:
Thanks David. I think you're making things very clear! 

Happy to help with drafting descriptions - I personally favour including them in the code so people are able to work with all the relevant information in the same place, but conscious that can make code quite long. 

Thanks for the other updates - completely understand about taking time. It's beyond my skills to incorporate them into the code otherwise I would offer. In the meantime, assuming that this controls the backlight: 

		if (i_counter*val_delay)>=backlight_off_delay and backlight_off_while_waiting == "yes" and (state != "play" and state != "pause"):
			mylcd.backlight (0)

And if it was amended to:

		if (i_counter*val_delay)>=backlight_off_delay and backlight_off_while_waiting == "yes" and (state != "play"):
			mylcd.backlight (0)

This would keep backlight on during play, but off when paused? 

Best,

Tom

davidfri, 2020-03-11:
Yes, it should ;)

SimonChelkowski, 2020-03-13:
Hi @davidfri,

I created the pull request #859. 

I implemented the following changes:
- implemented a change to date_and_time display option. In case n_cols=16 the year is displayed with only two digits to save space and make it work with the state_icons still
- correction of display of "Umlaut" characters (German language) for title, album and artist.
- changed defauls display settings such that date_and_time are displayed in row one, as it works nicely with the state_icons and does not result in a conflict if artist name is too long.

Lets see if it gets accepted :-)

Best regards,
Simon

mistersunshine2020, 2020-03-14:
Hi both, 

So I've upgraded my music player to accommodate a 4x20 display, as @davidfri convinced me of the benefits it could bring! It was a simple enough job to just transfer the connections from the 2x16 to the 4x20, so thanks again for all the work in helping get them connected in the first place. 

@SimonChelkowski I've not changed to your latest version in the pull request

I've got it set: 

Line 1: Time and Date (I agree that this is an ideal place for it with the state_icons
Line 2: Artist
Line 3: Title
Line 4: Track time and track number

A couple of observations: 

- I'm still getting some random characters appearing on the artist name, and it's the same issue I had with the original script that Simon submitted - if the new info doesn't entirely overwrite the previous info, random letters remain. Simon, how did you resolve that, and is it something that can be incorporated into the new screen? 
- On track time, I'm finding it's not syncing entirely when there are scrolling characters on any of the lines above. Is this based on the refresh rate of the characters? 

Amazing work as ever. I hope I'm helping rather than hindering with these comments. 

Best,

Tom

davidfri, 2020-03-14:
Hi Tom,
only a short answer to your problem:
I think the driver has a problem (or your display is to slow), so only the first X changes are made and then the buffer is full. 
Please go in the I2C_LCD_driver and set every delay to the double value. so 0.0001 -> 0.0002, etc.
The code only update the chars, which were changed, so in one cycle, only one char will be changed, in a different cycle 30 chars will be changed. 
On my tests I haven't an issue like yours. If the changes don't help, I will test more.
Is it only with the last version of the code or also with older ones?

closed 2020-05-15 by MiczFlor