MiczFlor/RPi-Jukebox-RFID #499: Sonos Play 1

Bartman83, 2019-02-24:
Can someone give me a hint, how I could connect the JukeBox4Kids with a Sonos Play 1. I would like to control music with the JukeBox4Kids, but the sound should come from sonos...

Ardios, 2019-02-28:
@Bartman83 

Do you want the sound output via Bluetooth?

Here is a link to the Wiki, there is a guide for it
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Bluetooth

Ardios

drocx, 2019-02-28:
> @Bartman83
> 
> Do you want the sound output via Bluetooth?
> 
> Here is a link to the Wiki, there is a guide for it
> https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Bluetooth
> 
> Ardios

Hello,

the Sonos Play 1 has no bluetooth.
But maybe it would be possible to remote control the spotify service on the sonos play 1 with rfid cards.

ddbkva, 2019-02-28:
perhaps you can try this 
https://www.netways.de/blog/2013/04/09/musik-ubers-netzwerk-mit-pulseaudio-und-mpd/
output via network for mpd player?

s-martin, 2024-05-10:
#627 is related